cnlohr/espusb #3: Error: unable to enumerate USB device

leopck, 2016-08-10:
Hi, I'm having slight trouble getting the espusb working on my NodeMCU ESP-12E. I think I have done something wrong but after troubleshooting this whole time, I think it's better for me to ask here.

I git pull your repo and change the SDK path then perform the usual "make" and "make burn" as shown in your video which compiles and flashes with ease and no error.

However, after wiring up my NodeMCU with USB Type B then connect to my Linux machines, these are the output I got from dmesg:

Wiring the ESP8266 to my Ubuntu 16.04 machine I get this messages:
[593134.728040] usb 3-1: new low-speed USB device number 5 using uhci_hcd
[593135.144039] usb 3-1: device not accepting address 5, error -71
[593135.144217] usb usb3-port1: unable to enumerate USB device

Wiring to Raspberry pi 3 (Raspbian Jessie):
[ 7814.928437] usb 1-1.4: new low-speed USB device number 72 using dwc_otg
[ 7815.348436] usb 1-1.4: device not accepting address 72, error -32
[ 7815.348535] usb 1-1-port4: unable to enumerate USB device

Wiring to OpenWRT 15.04:
[25945.270000] usb 1-1: new low-speed USB device number 6 using ehci-platform
[25945.750000] usb 1-1: device not accepting address 6, error -71
[25945.750000] usb usb1-port1: unable to enumerate USB device

My first schematics I added a 100ohm pull up resistor to D- (GPIO12) to 3.3v.

[image]

Then later I added a pull down resistor to D- and D+ but still no luck.

[image]

Did I do a mistake somewhere or missed out a step? Thanks again for this great project :+1:

cnlohr, 2016-08-10:
100 is def too beefy, try more than a 1k resistor.  Did the ESP boot?  Can you connect to it?

leopck, 2016-08-10:
The ESP booted up and I can see the serial console printing lots of "2222222222222222222222222". Only the dmesg of the USB (GPIO12&13) is showing this message. I'm currently trying your suggestion by swapping the 100ohm resistor with 1k ohm and 10kohm resistor.

Which circuit should I follow? the first or second(with pull-down resistor)?

On a side note: Don't you ever sleep? I see you committing and answering at every few hours haha

leopck, 2016-08-10:
Update:
I've swapped my pull-up resistor with 1k, 2.2k and 10k ohm all of them have the same result on all my machines (OpenWRT, Ubuntu and RPi3).

Sample output on serial console:
```
rl▒ldon't use rtc mem data
rl▒W▒▒▒266 usb driver
Loading Settings: af / 0 / 69 / 69
                                  Settings Loaded: ESP_A59278 / Default
                                                                       Opmode: 2
                                                                               Default SoftAP mode: "ESP_A59278":""
                                   Boot Ok.
                                           IGMP Joining: 0104a8c0 fb0000e0
                                                                          2222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222222
```

I plugged the ESP to the serial console while plugging the espusb pins to my Linux machine and the output is still showing error code -71 on ubuntu and openwrt while rpi3 shows error code -32.

Is there anything I missed out on the schematics? or is there somethg I need to enable first? because the serial console keeps printing "2222222" non-stop.

cnlohr, 2016-08-10:
(0) Please update to newest version - I just changed a bunch!

(1) Streams of 2 mean it's working.
(2) Should be following first schematic, with pull-up. 
(3) I sleep when I'm not in the middle of a project release... But I will be catching a few hours now.
(4) It can be really hard to debug without a logic analyser.  It is possible it could be failing somewhere down the line.  Try modprobing usbmon and send over a wireshark dump of the USB bus.  
(5) You can add more debugging information in the ESP by uncommenting 
```
#if 1
	printf( "(%08x)(%08x)\n", usb_internal_state.debug, usb_internal_state.eps[0].size_in  );
//	for( i = 0; i < 16; i++ )
//		printf( "%02x ", usb_internal_state.usb_buffer[i] );
//	printf( "\n" );
//	PIN_OUT_CLEAR = _BV(5);
#endif
```
in user_main.c

I hope we can get to the bottom of this for yours and anyone after your's sake!

leopck, 2016-08-21:
Hi cnlohr, sorry for the late reply, had to attend to real life.

I reattempted this again using the latest commits based on esp82xx, and I still encountered the same issues error code -71. So probably just to narrow down some things, I would like to verify a few things.

1. Does it work for you if you connect it to Windows 10 or Raspbian or Ubuntu? (This is for me to check if the issue is with my OS settings or my ESP8266)

2. Is there a way to get more debugging information from the ESP? Because the method stated in the previous comment doesn't seem to be inside the user_main.c anymore. Is there a way I can view more debugging information using your latest codes?

3. Assuming I can get my hands on a logic analyzer from my friend, is there any settings or tips that you can guide me with to know if the output is correct? Probably an image or the value I'm supposed to see from the logic analyzer.

Thanks for taking your time to help resolve this. I really would love to support this project and the esp82xx :D Can't wait to see more!

cnlohr, 2016-08-21:
I think debug code is still in user_main.c -- I posted it above.

But, I did move the GPIOs to GPIO4/5.  Also, a lot has changed in the last few days.  Check out a new copy.  You "should" be able to move the GPIOs back by updating the GPIO code in usb_config.h.

leopck, 2016-08-26:
OMG! I finally got some progress! After many trials and error, I think I got somewhere. So here's some update from me regarding this issue.

I've bought some new ESP8266 with breakout board (which just arrived) to test and it works with ESPUSB, so this got me to believe my NodeMCU is broken. However, my NodeMCU works perfectly for other applications and GPIO 4 and GPIO 5 works just fine as well. So I started trying multiple ways and replugging and resoldering my USB ports but still it doesn't work.

So, I took my Windows PC and used NodeMCU Flasher Programmer software and IT WORKS! 

I am not sure why this happens. Probably someone here can explain it? I am not sure whether am I the only one experiencing this issue (probably so since my NodeMCU is pretty old) but just in case, I'll leave some explanation on how I got it to work for others to reference.

On your Linux machine:
1. setup the usual
2. in espusb folder perform ```make all```
3. in web folder perform ```make``` and then ```Ctrl + C``` to interrupt the process to get the page.mpfs
4. transfer all files to Window machine:
```
image.elf-0x00000.bin
image.elf-0x40000.bin
page.mpfs
```
5.Download and extract NodeMCU Flasher
6.Follow my settings below

[image]

[image]

7.Flash to NodeMCU

After successfully making it work on NodeMCU, I retried on my Linux machine again but it doesn't work again. But I'm glad it started to work on my NodeMCU :D
Since it works temporarily using this method, not the most convenient method but since it works should we close this issue?

cnlohr, 2016-08-26:
I'd say, leave it open for now.  This is very strange and an actual problem.  Would you mind trying with different baud rates on Linux?  (Sorry, I am grasping for straws).

P.S. Pull the newest version and GUI.  I think you'll really like the demo :-D.  Be sure to connect to the wifi with a mobile device.

leopck, 2016-08-27:
Hi cnlohr, after some sleep I've found the issue. So according to esptool [https://github.com/themadinventor/esptool/tree/4deca7033e17d9b8d3b947a2d8f7ec3700d7d276#flash-mode---flash_mode--fm](url)

```Some ESP8266 modules, including the ESP-12E modules on some (not all) NodeMCU boards, are dual I/O and will only work with --flash_mode dio.```

So knowing this I referenced this issue to esp82xx with a PR to add ```-fm dio``` to the main.mf. I've pushed it to dev branch just in case something goes wrong. But after adding that -fm dio, the NodeMCU works well on Linux machine.

P.S. I've just checked the demo, it's awesome. Do you mind if I improved the web layout? Also, I realized I missed your live stream for espusb. How do I keep up to date to your live streams? Twitter?

cnlohr, 2016-08-27:
Well, last weekend was the first time I've ever tried one.  I may do another one next (not this!) weekend.

cnlohr, 2016-08-27:
But yes.  I suppose this is the best method to do it.  I am so glad con-f-use has been doing what he has been doing unifying everything!

closed 2016-08-27 by cnlohr

pastapojken, 2016-12-02:
Hi!
I am running in the same issue while using an esp-12f. Everything else works (AP, web page and all), but neither Windows nor Linux can recognize it no matter what i do...
As far as I know there isn't any significant difference between the 12e and the 12f, am I wrong?

leopck, 2016-12-02:
Hmmm, I don't have ESP-12F to test.
Changes made on ESP-12F:
- Breakthrough design, there are new breakthroughs, based on ESP-12E.
- The new four-layer board design.
- The new revision antenna RF performance optimization, communication distance is increased by 30% -50% compared to ESP-12E!
- Semi-hole chip technology, the whole IO leads, with metal shielding shell
- Passed FCC & CE & RoHS certification
- Built-board PCB antenna
- 4M bytes SPI Flash for programming.

Previously, my issue with the NodeMCU was that I couldn't flash properly because I didn't notice that there was a DIO and QIO flash version for ESP12E. 

Do you mind trying a few test for us? Can you build the espusb and flash using QIO settings? Currently by default I think it is DIO.

To do that just go to esp82xx/main.mf and edit this line:
```
else ifeq ($(CHIP), 8266)
burn : $(FW_FILE1) $(FW_FILE2)
        $(info NOTICE: Currently burning for ESP8266.  If you are on an '85 reconfigure CHIP.)
        ($(ESPTOOL_PY) $(FWBURNFLAGS) --port $(PORT) write_flash -fm dio $(FLASH_WRITE_FLAGS) 0x00000 $(FW_FILE1) 0x40000 $(FW_FILE2))||(true)
```
to 
```
else ifeq ($(CHIP), 8266)
burn : $(FW_FILE1) $(FW_FILE2)
        $(info NOTICE: Currently burning for ESP8266.  If you are on an '85 reconfigure CHIP.)
        ($(ESPTOOL_PY) $(FWBURNFLAGS) --port $(PORT) write_flash -fm qio $(FLASH_WRITE_FLAGS) 0x00000 $(FW_FILE1) 0x40000 $(FW_FILE2))||(true)
```

Let us know if that fix up the issue.

T-vK, 2016-12-02:
I'd say it's probably a wiring issue. A photo or something would probably help.

pastapojken, 2016-12-02:
Tried QIO, unfortunately it didn't change anything... :/

[image]

My wiring is as shown, red is 3.3v, orange is 5v, black is GND, white is D- and green D+.
VCC, EN, GPIO2 and GPIO0 are connected to 3.3, while GND and GPIO15 are connected to GND.
GPIO4 is connected to D+ and a 1k resistor that goes to 3.3v.
GPIO5 is connected to D-.
I should admit I am a bit of a newbie when it comes to the esp, so I'm not sure why pin 0 and 2 are connected to 3.3v, but if they aren't the esp doesn't start...
Anyway, hopefully this helps, thanks!

cnlohr, 2016-12-02:
(1) You are correct about the GP0 and GP2, they, in conjunction with GP15 control boot mode.  If you need to use those pins, you can simply use a resistor to pull them up/down.  There is actually concern if you don't.  Because GPIO2 is a status bit used for controlling the LED status, it may misbehave if forced high.  Consider wiring GP2 up to VCC with through a 1k resistor.

(2) D- is GPIO4, your color, White indicates this correctly, your statement is backwards.  The resistor being on D- is also correct.

(3) Do you have any way of looking at the serial console?  Just to make sure the ESP is booting with basic functionality? 

(4) Can you connect to the ESP wirelessly?

(5) @leopck  -- I thought it was DOUT/QOUT?  Gah, I always get those flash modes confused!  Normally, you can use the 8285 flash mode with the 8266, but it doesn't buy you GPIO 8/9.  Can only use those on the actual 8285.

I'd love to get to the bottom of this!

pastapojken, 2016-12-02:
...fixed it! I had a bad connection on D-... Resoldered and it all works, sorry and thanks for the help!

cnlohr, 2016-12-02:
awesome!!!  DEVELOP HAPPY!

leopck, 2016-12-03:
@cnlohr haha! I also constantly get confused over the flash modes! I think the confusion was due to us discussing over this issue and the other issue on the ESP8285 that you tried to flash over the net. ESP8285 only supported DOUT while some of ESP-12E supported only DIO and the rest of the ESPs supported QIO/DIO/DOUT/QOUT. So I guess we're always mixing them up since it's all over. Not sure about ESP32 though. I've been spending a lot of my time on ESP32 nowadays.

cnlohr, 2016-12-03:
I just started devving for one right now.  The I2S bus's raw clock rate is slower from what I can see (so no crazy RF hax) but the fact it can read 16 bits simultaneously at 40 MHz is pretty rockin'

leopck, 2016-12-03:
Wow, you sure work fast. I just received my LAN8720 and DP83848 MAC Ethernet for when ESP32 releases the API for Ethernet and I see that ESP32 now is porting Bluedroid for Bluetooth! 

Bluedroid is an open source Bluetooth stack, nRF and TI's stacks are not open source. Might be worth looking into that. I'm spending my time going thru the Bluedroid's stack for the moment. Hopefully they release a document to simplify this process.

[https://github.com/espressif/esp-idf/tree/master/components/bt/bluedroid](https://github.com/espressif/esp-idf/tree/master/components/bt/bluedroid)

cnlohr, 2016-12-03:
nRF is sooo simple.  That's unfortunate.  I hope they open up the wifi stack like they did on the '8266.  Sending packets around at 72 Mbits/s broadcast in a connection-less, association-less protocol was SUPER RAD!

leopck, 2016-12-03:
@cnlohr Yes! I know right! nRF did a wonderful job with their Bluetooth API, I was kind of addicted on nRF boards before ESP32 was announced.

WiFi stack? You mean the ESP32's WiFi stack is closed sourced?

cnlohr, 2016-12-03:
It depends on what part you're talking about, but the PHY side of their stack is closed :(