cnlohr/espusb #4: Hardware design has outdated and missing libraries

con-f-use, 2016-08-12:
I don't know if you noticed, but the hardware design doesn't work with recent versions of KiCad. 

First of all a few [custom libraries](http://pastebin.ubuntu.com/23049181/), you used were not exported with the project. 

Second, KiCad broke compatibility of standard footprints about a year ago. Resistors and such are now much smaller in EESchema, leading to broken connections in the schematic, because KiCad doesn't automatically restore connections. It's stupid, but that's the way it is.

[image]

cnlohr, 2016-08-12:
I will fix this tonight.  But, keep in mind that it is an untested design.  I should include schematics of what I am actually working with, I guess.

cnlohr, 2016-08-13:
...My -cache file does include even things like resistors.  I'm not sure why your kicad is preferring your personal libraries.  Also, that's really annoying since I used the libraries in kicad!

con-f-use, 2016-08-13:
Yes, it is annoying, but not a problem. The latest dev version of KiCad seems to have a dialog that asks about it. After that, your schematic works fine. Nonetheless, I find the library management in KiCad a bit odd. If you break compatibility with your old standard library, your program should gracefully port old projects to the new library without much ado, and do it from the start.

cnlohr, 2016-08-13:
I thought it did!  That was the whole point of the -cache files.

con-f-use, 2016-08-13:
Didn't mean to imply anything on your side. It's one of many known problem with KiCad, but ever since version 4 and CERN contributing, problems keep disappearing.

cnlohr, 2016-08-16:
Hmm, I appreciate the updating and cleanups - but - one of the biggest problems is going to be keeping these things in sync.  I'm not really sure how to do that.  ESPUSB, ColorChord and esp8266ws2812i2s somehow need to be kept in sync.  Also, still haven't gotten a chance to upgrade kicad, will be trying tonight or tomorrow.  I spent so much time this weekend getting the ColorChord triple-header done.  But, back to work now.

con-f-use, 2016-08-16:
The easiest way to keep them in sync would be to create a repository for the common files and then include that as a submodule in all the derivative ESP projects. As bonus it would be really easy to start a new esp-based project.

cnlohr, 2016-08-17:
I will pick up in that thread.

cnlohr, 2016-09-04:
I updated the design in github to reflect reality.  Also, you can use the design in esp82xx for USB as well.  Hmm... We are going to need to make that bootloader sooner or later.

con-f-use, 2016-09-04:
Bootloader? What? Why?

cnlohr, 2016-09-04:
Oh!  Bootloader = we need something that can reside in the first 4k? of memory, that will boot.  If plugged into USB gives you a brief period of time before booting the main program, or, if not plugged into usb immediately boots the program.  This makes it possible to completely remove the serial programming headers, or re-appropriate the serial port for other use.

con-f-use, 2016-09-04:
Ah, you want an USB bootloader. Didn't make the connection, because board layout here...
Anyway, we basically have a bootloader over network already, don't we? I men you can update the firmware via net-flashing. Is it even worth the effort, seeing users will always need the serial port for flashing whatever bootload? I mean it's a nice option to have, but I'm pretty content with net-flashing.

cnlohr, 2016-09-04:
The big distinction is with the fact the usb bootloader would need to be "unbrickable" whereas netflashing and current usb flashing.  

Plus... what in the world is that board for?  Looks like WS2812B's... Is that an ESP-01?... but I am confused.

con-f-use, 2016-09-04:
Why not be ambitious and make the netflash unbrickable ^^

>  Looks like WS2812B's... Is that an ESP-01?

Correct. It's for a prop I'm making with a friend. Basically something like [this](https://i.ytimg.com/vi/sAENTUJPbN8/hqdefault.jpg), with the stone lit and larger. I got fifty ESP-01 for only $10 shipping. That's how I came here in the first place.

Def152, 2016-09-04:
> fifty ESP-01 for only $10 shipping
How is it possible :P? You work for espressif? Aliexpress? Ebay? Can you share your secret?

con-f-use, 2016-09-04:
No secret, someone wanted to be rid of them. I was first to reply. Bham, just had to pay shipping. One time deal, I'm afraid. :-(