cnlohr/espusb #5: Document how to build this

probonopd, 2016-08-12:
Would be good to have some instructions about the required build environment, e.g., which compilers, which SDKs, etc.

leopck, 2016-08-12:
Hey I think I can help answer this, I got it to work by installing the esp-open-sdk and esp-iot-sdk_v1.5.1. You can follow the installation instructions from this link: [https://github.com/pfalcon/esp-open-sdk](https://github.com/pfalcon/esp-open-sdk) 
In fact, my instructions below is just from their website, except a few addition below.

I used Ubuntu 16.04, so the installation
```
$ sudo apt-get install make unrar autoconf automake libtool gcc g++ gperf \
    flex bison texinfo gawk ncurses-dev libexpat-dev python-dev python python-serial \
    sed git unzip bash help2man wget bzip2
```
Later Debian/Ubuntu versions may require:
```
$ sudo apt-get install libtool-bin
```

Create an esp8266 folder, just to be neat:
```
mkdir ~/esp8266/
cd ~/esp8266/
```

Git clone the esp-open-sdk and make it:
```
git clone --recursive https://github.com/pfalcon/esp-open-sdk.git
cd esp-open-sdk/
make
cd ..
```

Download the ESP8266 nonos sdk v1.5.1 from espressif:
```
wget --content-disposition "http://bbs.espressif.com/download/file.php?id=1046"
unzip ESP8266_NONOS_SDK_V1.5.1_16_01_08.zip
esp_iot_sdk_v1.5.1/
```

Git clone espusb to esp8266 folder and then modify the Makefile to your configuration:
```
git clone https://github.com/cnlohr/espusb.git
cd espusb
vi Makefile
```
Find and change the following configuration to your settings:
```
GCC_FOLDER:=~/esp8266/esp-open-sdk/xtensa-lx106-elf
ESPTOOL_PY:=~/esp8266/esp-open-sdk/esptool/esptool.py
SDK:=~/esp8266/esp_iot_sdk_v1.5.1/esp_iot_sdk_v1.5.1
PORT:=/dev/ttyUSB0 #Location of your ESP8266
```
Save & Quit

Make espusb and flash it to ESP8266:
```
make
make burn
```
Done, it should compile and burn into ESP8266 nicely.

As for hardware(Low speed configuration, 1.5Mbits)

[image]

cnlohr is working hard now to enable USB Full speed implementation (12 MBit/s), the hardware will be slightly different.

Hope this helps :) Happy hacking.

P.S I haven't tested on other versions of esp_iot_sdk v1.5.1 and v2.0 but I didn't succeed in compiling in v2.0 probably if you succeed, you can share how you did it here :D

probonopd, 2016-08-12:
Thank you very much!

closed 2016-08-12 by probonopd