cnlohr/espusb #8: Error on esptool.py: A fatal error occurred: Invalid head of packet ('\xe0')

leopck, 2016-08-21:
I just pulled your latest commit and make the application, it seems that I'm having this issue on my Ubuntu 16.04 machine to burn into my NodeMCU
```
esptool.py v1.2-dev
Connecting...
Running Cesanta flasher stub...

A fatal error occurred: Invalid head of packet ('\xe0')
```

I checked the user.cfg and found the issue (for me at least) to be ```FWBURNFLAGS  = -b 1000000``` after changing this to ```FWBURNFLAGS  = -b 115200``` or ````FWBURNFLAGS  = -b 1500000``` I can burn into my NodeMCU nicely.

Does ```FWBURNFLAGS  = -b 1000000``` work for you?

and saw the esp82xx really great job there!

cnlohr, 2016-08-21:
-b 1000000 works for me, but so does 1,500,000.  Perhaps we should leave this issue here for others to chime in!  

I know I'm weird.  I use $0.90 USB to serial adapters..  http://www.electrodragon.com/product/usb-ttl-serial-ch340-board/

leopck, 2016-08-24:
So here's some update, I took some of the USB-Serial I have around and tested and you're right. Somehow it works very nicely with CH340, PL-2303HX and FTDI232 however, testing again using the CP2102 doesn't work and gives that error ```Invalid head of packet ('\xe0') ``` hence, the NodeMCU that is also based on CP2102 is also showing this error.

Not a big issue, just change the ```FWBURNFLAGS = -b 115200``` or ```FWBURNFLAGS = -b 1500000```. If someone else can verify this then we can close this.

john7doe, 2016-11-17:
I can confirm that changing the setting to  115200 helped me. Using a CP2102 based dongle on osx 10.11.3

cnlohr, 2016-11-23:
I think I'll leave this issue open forever :-p

umotesla, 2017-06-13:
I can confirm this always happens with CP2102, while perfectly working with FTDI

cnlohr, 2017-06-14:
Yep. 1500000 vs 1000000 vs 2000000 baud.  When you get the crazy high speeds, it is device-dependent.

alkopop79, 2019-01-08:
Where can I find the user.cfg file on Mac OS X?

elasticdotventures, 2019-01-16:
Just wanted to chime in that by further reducing the baud rate from 115200 to 57600 it solved a similar issue:

Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 80:7d:3a:f3:37:8c
Uploading stub...

A fatal error occurred: Invalid head of packet (0x65)

gardendave, 2019-01-20:
thanks for leaving this open.  
115200 worked for me.

FlackMonkey, 2019-03-06:
Confirm to have issue, and solved by 115200 flashing speed  for ESP32 (TTGO mini T7 v1.3)

reppr, 2019-04-16:
Same issue on Geekworm KE-WROVER-C20 EASY KIT ESP32_WROVER 8MB PSRAM
fiddling with the boot button (labeled "USER" on this board) did not help
upload speed 115200 works well

leon-v, 2020-11-28:
> Just wanted to chime in that by further reducing the baud rate from 115200 to 57600 it solved a similar issue:
> 
> Chip is ESP32D0WDQ6 (revision 1)
> Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
> MAC: xx:xx:xx:xx:xx
> Uploading stub...
> 
> A fatal error occurred: Invalid head of packet (0x65)

In my case, the devkit was expecting 115200 and the idf was set to 460800

cnlohr, 2020-11-28:
It is very unusual that 460800 is outside of a viable upload speed, but, with weird USB-serial chips that makes some sense.

cnlohr, 2020-11-28:
Also, I am going to close this issue, since it is basically resolved for anyone who would care, and does not reflect an issue with the codebase.

closed 2020-11-28 by cnlohr