cnlohr/espusb #10: A little how to use section in the readme would be nice

T-vK, 2016-09-30:
I have a very hard time understanding how to use your project.  
I just want to make my ESP8266 act as a USB keyboard. 
And the file `user/usb_config.h` makes me think that the implementation for this does exist.  

It would be really nice if you could add a little section to the readme explaining what user.cfg is all about and what to do in order to get the ESP8266 to act as the desired device.  

At the moment I'm just very confused. I mean what do the variables in user.cfg even do? If the `IP` for example doesn't set the IP address for the ESP8266, then what does it do? 

Best regards
tvk

PS: I love your work!

cnlohr, 2016-10-01:
Hmm... It does already map as a keyboard + mouse.  You would have to modify it to say it only has one interface and select just the keyboard interface to make it keyboard only.  To really "get" user/usb_config.h you'll have to read up a good bit about USB descriptors.

The IP parameter sets the parameter for where the netburn system expects the ESP8266.  You use the web-based configuration to set the ESP's network information, i.e. if it is a softap or if it is connected to an infrastructure.

T-vK, 2016-10-04:
Oh, I didn't realize that it would map as keyboard+mouse by default. I thought this was just a general USB library. It might be a good idea to put that information into the readme file.

And about netburn: How do I make my esp8266 work with that? It's currently running nodemcu, I don't think it understands netburn. Do I have to do a netburn or can I just do a burn to a serial port?

cnlohr, 2016-10-04:
Netburn only works with esp82xx projects, once you serial-burn the firmware once, you can netburn every time thereafter.

T-vK, 2016-10-16:
I have tried to get it to work for over a week now and I'm starting to think that you use something different or maybe an outdated version?  

I successfully got the toolchain to work within a docker image: https://github.com/T-vK/docker-esp-open-sdk/blob/master/Dockerfile But that's only a build environment, I have no idea how I would pass a serial port through to the image.  
So I tried to install the toolchain in a VM following this guide: https://github.com/esp8266/esp8266-wiki/wiki/Toolchain But sadly the guide is outdated and wasted hours of my time. 

So I digged up the official documentation that espressif provides and found that they provide a VM image that has the toolchain preinstaled (http://bbs.espressif.com/download/file.php?id=1450). I just had to add the SDK to `~/Share/ESP8266_NONOS_SDK`. 

Now here is what I did:

```
git clone https://github.com/cnlohr/espusb.git
cd espusb
git submodule update --init --recursive
# adjusted these values in the user.cfg
# SDK_DEFAULT  = $(HOME)/Share/ESP8266_NONOS_SDK
# ESP_GCC_VERS = 4.8.2
make burn
```

This fails:
```
fails esp82xx/common.mf:37: Warning: No shell variable 'ESP_ROOT', using '/home/esp8266/esp8266/esp-open-sdk'
Makefile:10: warning: overriding commands for target `usbburn'
esp82xx/main.mf:113: warning: ignoring old commands for target `usbburn'
...
make: /home/esp8266/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc: Command not found
make: *** [image.elf] Error 127
```

So I set the shell variable for the command by running `env ESP_ROOT="~/Share/ESP8266_NONOS_SDK" make burn`.  
Result:  
```
Makefile:10: warning: overriding commands for target `usbburn'
esp82xx/main.mf:113: warning: ignoring old commands for target `usbburn'
...
/home/esp8266/Share/ESP8266_NONOS_SDK/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc: not found
make: *** [image.elf] Error 127
```
Okay, now that is really weird. Accoring to espressif the xtensa-lx106-elf should be in /opt. And sure enough it was preinstalled in /opt/xtensa-lx106-elf.  
But I was curious and pasted a copy of that folder into my ESP8266_NONOS_SDK folder and ran the same command again:  

```
Makefile:10: warning: overriding commands for target `usbburn'
esp82xx/main.mf:113: warning: ignoring old commands for target `usbburn'
...
user/user_main.c:10:24: fatal error: ws2812_i2s.h: No such file or directory
 #include "ws2812_i2s.h"
                        ^
compilation terminated.
make: *** [image.elf] Error 1
```

What is wrong here? There are so many issues. How hard can it be to get this to work?

cnlohr, 2016-10-17:
Whoops!  Forgot to take out the refernce.  git pull.  Try agian.

T-vK, 2016-10-17:
Still not working. :/ 
I'll paste the complete terminal input/output: 

```
esp8266@esp8266-VirtualBox:~/espusb$ git pull
remote: Counting objects: 4, done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 4 (delta 0), reused 0 (delta 0), pack-reused 0
Unpacking objects: 100% (4/4), done.
From https://github.com/cnlohr/espusb
   c761d8b..fe27767  master     -> origin/master
Updating c761d8b..fe27767
Fast-forward
 user/user_main.c | 3 ---
 1 file changed, 3 deletions(-)

```

```
esp8266@esp8266-VirtualBox:~/espusb$ sudo cp -r /opt/xtensa-lx106-elf ~/Share/ESP8266_NONOS_SDK/
esp8266@esp8266-VirtualBox:~/espusb$ env ESP_ROOT="~/Share/ESP8266_NONOS_SDK" make burn
Makefile:10: warning: overriding commands for target `usbburn'
esp82xx/main.mf:113: warning: ignoring old commands for target `usbburn'
~/Share/ESP8266_NONOS_SDK/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -mlongcalls -Os -I/home/esp8266/Share/ESP8266_NONOS_SDK/include -Iesp82xx/include -I. -Iesp82xx/fwsrc -Iuser -DICACHE_FLASH -DWEB_PORT=80 -DCOM_PORT=7777 -DBACKEND_PORT=7878  -DSLOWTICK_MS=100 -DVERSSTR='"Version: fe277-dev - Build Mon, 17 Oct 2016 13:15:09 +0800 with -DICACHE_FLASH -DWEB_PORT=80 -DCOM_PORT=7777 -DBACKEND_PORT=7878  -DSLOWTICK_MS=100"'  esp82xx/fwsrc/uart.c esp82xx/fwsrc/esp82xxutil.c esp82xx/fwsrc/flash_rewriter.c esp82xx/fwsrc/http.c esp82xx/fwsrc/commonservices.c esp82xx/fwsrc/http_custom.c esp82xx/fwsrc/mdns.c esp82xx/fwsrc/mfs.c user/custom_commands.c user/user_main.c user/usb_asm_1bit.S user/usb.c -flto -Wl,--relax -Wl,--gc-sections -nostdlib -L/home/esp8266/Share/ESP8266_NONOS_SDK/lib -L~/Share/ESP8266_NONOS_SDK/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.2/libgcc.a /home/esp8266/Share/ESP8266_NONOS_SDK/lib/libmain.a /home/esp8266/Share/ESP8266_NONOS_SDK/lib/liblwip.a /home/esp8266/Share/ESP8266_NONOS_SDK/lib/libssl.a /home/esp8266/Share/ESP8266_NONOS_SDK/lib/libupgrade.a /home/esp8266/Share/ESP8266_NONOS_SDK/lib/libnet80211.a /home/esp8266/Share/ESP8266_NONOS_SDK/lib/libwpa.a /home/esp8266/Share/ESP8266_NONOS_SDK/lib/libphy.a /home/esp8266/Share/ESP8266_NONOS_SDK/lib/liblwip.a /home/esp8266/Share/ESP8266_NONOS_SDK/lib/libcrypto.a /home/esp8266/Share/ESP8266_NONOS_SDK/lib/libpp.a ~/Share/ESP8266_NONOS_SDK/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.2/libgcc.a -T /home/esp8266/Share/ESP8266_NONOS_SDK/ld/eagle.app.v6.ld -B/home/esp8266/Share/ESP8266_NONOS_SDK/lib  -o image.elf
/mnt/Share/ESP8266_NONOS_SDK/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: image.elf section `.text' will not fit in region `iram1_0_seg'
/mnt/Share/ESP8266_NONOS_SDK/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: region `iram1_0_seg' overflowed by 728 bytes
collect2: error: ld returned 1 exit status
make: *** [image.elf] Error 1
esp8266@esp8266-VirtualBox:~/espusb$ 
```

cnlohr, 2016-10-18:
Yes, looks like SDK version is too new.  I think it must be 1.5.2 or older.  See what version is currently in use in the ESP8266_NONOS_SDK, and make sure to download 1.5.2 or older.

leopck, 2016-10-18:
@cnlohr SDK 1.5.4 works too. Haven't got the SDK 2.0 to work though. https://github.com/cnlohr/espusb/issues/5

T-vK, 2016-10-18:
Yeah, I'm using 2.0... @cnlohr Any idea when you'll upgrade to 2.0?

leopck, 2016-10-18:
@T-vK Hi you can refer to this issue on document about this:
https://github.com/cnlohr/espusb/issues/5
I added some notes below of my comment.

T-vK, 2016-10-21:
@leopck Thank you. That helped a lot! Remember that esp-open-sdk already takes care of downloading the sdk. That makes your wget redundant.  

For anyone reading this who doesn't feel like bloating your system with all these dependencies. I created an executable docker image that can easily compile and flash your stuff.  https://github.com/T-vK/docker-esp-sdk  
It is very well documented. :)  
You can also just refer to my Dockerfile if you want to know all steps necessary to get the build environment running. https://github.com/T-vK/docker-esp-sdk/blob/master/Dockerfile

closed 2016-12-07 by leopck