cnlohr/espusb #11: GPIO pair for D- and D+

mikewen, 2016-10-02:
Can you explain a bit on 'any GPIO pair may be used. It is important to note that both D- and D+ MUST be adjacent.' ?

Is it means GPIO12 and GPIO13 (D-, D+) will work, but GPIO12 and GPIO14 (D-, D+) won't ?

Why is that? 

Thanks a lot! This is great project!

cnlohr, 2016-10-03:
Correct!  

That is because I mask off both bits using the extui command, so, you can't select the bits separately.  Admittedly at this point, for low-speed only, this could be re-worked... but that's how it's setup now.

atc1441, 2017-02-23:
Hey where can i change the usb pins in the source code? 
i want to use rxd and txd with an esp-01 module

T-vK, 2017-02-23:
That would be in [usb_config.h](https://github.com/cnlohr/espusb/blob/master/user/usb_config.h) I think.

atc1441, 2017-02-23:
I was loking there and there is this define: PERIPHS_IO_MUX_GPIO5_U and FUNC_GPIO4
  if i change this to 3 or 1 witch is the rxd and txd, it shows me errors

T-vK, 2017-02-23:
I don't know. Try PERIPHS_IO_MUX_U0RXD_U, PERIPHS_IO_MUX_U0TXD_U, FUNC_GPIO3 and FUNC_GPIO1

atc1441, 2017-02-23:
Thank you it did compile now. But did not work for me i will try it with adjacent pins like gpio2 or gpio0 maybe it will not disturb the esp bootloader

atc1441, 2017-02-24:
i did it now with the original pins gpio 4 and 5 see here: [http://atcnetz.blogspot.de/](http://atcnetz.blogspot.de/)

it work great as a mini usb stick :)

thank you cnlohr and T-vK

cnlohr, 2017-02-24:
Spectacular.   

In retrospect, I could have engineered it in such a way that it would not require adjacent pins.  I wish I had done that, but that ship has sailed :(

anishsane, 2024-08-06:
Do you know if we can use the GPIO0/1/2 pins?
They are used to enter bootloader mode etc. They must not be pulled low at the time of power on.

I am not trying to nitpick. I have a few tiny ESP-01 modules that only have GPIO0..3 exposed as pins on the header.

cnlohr, 2024-08-15:
I honestly don't know.  I am unaware of any issues.  Give it a shot and report back?