cnlohr/espusb #12: Joystick over ESPUSB

leopck, 2016-10-03:
I've been trying to enable joystick over espusb, I've managed to add the joystick hid descriptor based off a joystick I have (Microsoft Force Feedback Joystick 2). I am trying to attempt to create a composite device with mouse + keyboard + joystick... is this possible?

Anyway, I managed to get some readings from the 'dmesg' by adding the hid descriptor into the usb_config.h but I'm receiving lots of errors as well :( I think I'm missing something or doing something wrong. Any advice? I've placed the usb_config.h together with this post.

```
[   28.127483] hid-generic 0003:ABCD:8266.0008: input,hidraw7: USB HID v1.10 Joystick [CNLohr ESPUSB2] on usb-0000:00:1d.0-1.2/input2
```

[image]

joystick_hid_desc
```
//From Microsoft Force Feedback Joystick 2 Reverse Engineered
static const uint8_t joystick_hid_desc[61] = {   /* USB report descriptor */
    0x05, 0x01,                    // USAGE_PAGE (Generic Desktop)
    0x09, 0x04,                    // USAGE (Joystick)
    0xa1, 0x01,                    // COLLECTION (Application)
    0x85, 0x01,                    //   REPORT_ID (1)
    0xa1, 0x02,                    //   COLLECTION (Logical)
    0x09, 0x32,                    //     USAGE (Z)
    0x09, 0x31,                    //     USAGE (Y)
    0x09, 0x30,                    //     USAGE (X)
    0x15, 0x00,                    //     LOGICAL_MINIMUM (0)
    0x26, 0xff, 0x00,              //     LOGICAL_MAXIMUM (255)
    0x35, 0x00,                    //     PHYSICAL_MINIMUM (0)
    0x46, 0xff, 0x00,              //     PHYSICAL_MAXIMUM (255)
    0x75, 0x08,                    //     REPORT_SIZE (8)
    0x95, 0x03,                    //     REPORT_COUNT (3)
    0x81, 0x02,                    //     INPUT (Data,Var,Abs)
    0xc0,                          //   END_COLLECTION
    
    0x85, 0x02,                    // REPORT_ID (2)
    
    0xa1, 0x02,                    //   COLLECTION (Logical)
    0x05, 0x09,                    //     USAGE_PAGE (Button)
    0x29, 0x02,                    //     USAGE_MAXIMUM (Button 2)
    0x19, 0x01,                    //     USAGE_MINIMUM (Button 1)
    0x95, 0x02,                    //     REPORT_COUNT (2)
    0x75, 0x01,                    //     REPORT_SIZE (1)
    0x25, 0x01,                    //     LOGICAL_MAXIMUM (1)
    0x15, 0x00,                    //     LOGICAL_MINIMUM (0)
    0x81, 0x02,                    //     Input (Data, Variable, Absolute)
    0x95, 0x01,                    //     Report Count (1)
    0x75, 0x06,                    //     Report Size (6)
    0x81, 0x01,                    //     Input (Constant) for padding    
    0xc0,                          //   END_COLLECTION
0xc0 // END_COLLECTION
};
```

config_descriptor
```
	//Interface Descriptor:: Joystick
	0x09,//	bLength
	0x04,//	bDescriptorType
	
	//Need to adjust this accordingly if we add more devices :: bInterfaceNumber is like an ID so must be unique
	0x02,//	bInterfaceNumber
	0x00,//	bAlternateSetting
	0x02,//	bNumEndPoints
	0x03,//	bInterfaceClass   (Human Interface Device Class)
	0x00,//	bInterfaceSubClass   
	0x00,//	bInterfaceProtocol   
	0x00,//	iInterface
	
	
	//HID Descriptor
	0x09,//	bLength
	0x21,//	bDescriptorType
	0x10,0x01,//	bcdHID
	0x00,//	bCountryCode
	0x01,//	bNumDescriptors
	0x22,//	bDescriptorType   (Report descriptor)
//	0x3F,0x05,//	bDescriptorLength
	61, 0x00,// bDescriptorLength
	
	//Endpoint Descriptor IN
	0x07,//	bLength
	0x05,//	bDescriptorType
	0x81,//	bEndpointAddress   (IN Endpoint)
	0x03,//	bmAttributes	(Transfer: Interrupt / Synch: None / Usage: Data)
	0x40,0x00,//	wMaxPacketSize   (64 Bytes) 
	0x01,//	bInterval
	
	
	//Endpoint Descriptor OUT	
	0x07,//	bLength
	0x05,//	bDescriptorType
	0x02,//	bEndpointAddress   (OUT Endpoint)
	0x03,//	bmAttributes	(Transfer: Interrupt / Synch: None / Usage: Data)
	0x10,0x00,//	wMaxPacketSize   (16 Bytes) 
	0x04,//	bInterval
```

descriptor_list[] (honestly, I don't have a clue on the 0x2200 thing, so i just added and increased the number sequentially :D)
```
descriptor_list[] = {
	{0x0100, 0x0000, device_descriptor, sizeof(device_descriptor)},
	{0x0200, 0x0000, config_descriptor, sizeof(config_descriptor)},
	{0x2200, 0x0000, mouse_hid_desc, sizeof(mouse_hid_desc)},
	{0x2200, 0x0001, keyboard_hid_desc, sizeof(keyboard_hid_desc)},
	{0x2200, 0x0002, joystick_hid_desc, sizeof(joystick_hid_desc)}, <---- added
	{0x0300, 0x0000, (const uint8_t *)&string0, 4},
	{0x0301, 0x0409, (const uint8_t *)&string1, sizeof(STR_MANUFACTURER)},
	{0x0302, 0x0409, (const uint8_t *)&string2, sizeof(STR_PRODUCT)},	
	{0x0303, 0x0409, (const uint8_t *)&string3, sizeof(STR_SERIAL)}
};
```

leopck, 2016-10-03:
Oops, looks like I can't upload .h files here... so i renamed the file format to .txt pls change this to .h again

[usb_config.txt](https://github.com/cnlohr/espusb/files/506769/usb_config.txt)

cnlohr, 2016-10-03:
set this to three: ```0x02, // bNumInterfaces (Normally 1)```

Actually, it looks like other work needs to be done.  I am not the best at USB descriptors, but this doesn't look right.   Also, be sure to update all total lengths that you've changed, i.e. 

```0x5A, 0x00, // wTotalLength ```