cnlohr/espusb #14: Is espusb compatible with FreeRTOS version?

leopck, 2016-10-08:
I'm curious is the espusb project compatible with the FreeRTOS version of the espressif SDK? I was curious because I heard that the ESP32 is based on the FreeRTOS version. I took a very very short glance at the ESP32 codes and it looks pretty much the same on the C side at the least to ESP8266.

It'll be great if this works on ESP32 as well :D

cnlohr, 2016-10-08:
It certainly does not work on the ESP32 right now, but considering it now works on no SDK, it should also work with FreeRTOS.    If you don't mind porting and submitting changes to get it to compile that would be awesome.

leopck, 2016-10-08:
Awesome, my ESP32 devkitC should arrive some time this week. I'll keep you posted on this once I receive the ESP32 and I'll give the porting a try as well

andyhu, 2017-03-23:
Any update on this?

cnlohr, 2017-03-23:
"compatible" yes, however, only limitedly so.  You would need to set up the interrupts directly, but other than that, it should "just work."

andyhu, 2017-03-23:
Really hope it could work on ESP32, WiFi + BLE + USB, what a combination!

cnlohr, 2017-03-23:
Maaan... so... I think the ESP32 is actually capable of doing full-speed USB (serial, network, etc. support), but I'm just spending all my free time now on libsurvive.

ConstantinoSchillebeeckx, 2017-08-09:
any movement on this? I'm also looking to use this with the ESP32! 🌟

cnlohr, 2017-09-09:
So, the big reason I have been at a standstill on that is because I really wanted to do full-speed.  But it's really hard to do full-speed.  I don't actually know if it's even possible.

Is it worth the effort to do low-speed?  I don't think so, but if people feel differently... It would be rather easy to port.

sbuller, 2017-09-09:
I have interest in using the ESP32 as a keyboard controller. As far as I'm aware (so very not an expert...), there's no advantage to full-speed for a keyboard.

cnlohr, 2017-09-09:
That is correct.  HID devices need only low-speed.  Hnggg maybe it is worth the 10 or so hours it would take.

idesignstuff, 2017-09-09:
Oh man. I think a lot of people would use it. I'd love having the BT/WiFi
interface and flash storage that the pro micro is missing.

Christian

cnlohr, 2017-09-09:
To do something like that, i.e. talk to flash sticks, etc.  You would need full-speed :(

idesignstuff, 2017-09-09:
Sorry, I wasn't clear.
For a project I'm starting on, I'm using a pro micro to make a hot key
panel, for sketchbook pro. For instance, canvas rotate, undo, brush size,
brush opacity, color picker, etc.

I'd like to make it easy for the user to change what message is sent when
hotkeys are pressed, but the pro micro is limited. If I had a WiFi or BT
interface for setting up hotkey assignment, and saving settings to flash,
with HID over USB to send keystrokes or key combos or strings, that would
be ideal.

I'm just saying that, right now, I have an application that would benefit
hugely by enabling hid on an esp32 Dev board.

Christian

blackketter, 2017-09-09:
+1 for low-speed support!  I'm primarily interested in HID, but it would also be super useful for a serial console and firmware updates.

I'm further imagining that once the USB support is reliable, ESP32 dev boards could be even cheaper without a hardware USB chip.

ConstantinoSchillebeeckx, 2017-09-09:
my application is using an[ ESP32 as a keyboard](https://github.com/ConstantinoSchillebeeckx/tacocat-keyboard), so another +1 for low-speed support!

cnlohr, 2017-09-11:
I know Sprite was considering using the RMT module for low speed.  I will be pinging him on Twitter to see where he ended up.  It would be much better than the way it is done in ESPUSB, since it wouldn't require blocking a core while receiving.   Perhaps I could take his start and put it in to finish.

benjaminaigner, 2017-11-13:
@idesignstuff 
Might be perfect for you:
github.com/asterics/FABI

We also want to get rid of the AVR Mega32U4 and do everything in the ESP32 -> one more wish for low speed USB on the ESP32 :-)

@cnlohr 
I admire your work with ESPUSB, really nice stuff! And really nice working hours (according to the forum posts), Saturday evening seems to be a good hacking time :-).

cnlohr, 2017-11-15:
Last weekend I've started looking into using the ULP for low-speed USB, however, it would require the ULP being clocked by another source, and I'm unsure if that's possible.

bryantfhayes, 2020-07-29:
Is this still being developed?

dzcpy, 2020-08-04:
@bryantfhayes You may try ESP-12K, it supports USB natively

cnlohr, 2020-08-12:
@bryantfhayes I don't think anyone is actively developing this, though it is likely to be ported to a modern esp82xx soon.

Closing issue because there is no plan to port to a freertos platform.

closed 2020-08-12 by cnlohr