cnlohr/espusb #15: How to Send key press to host?

mikewen, 2016-10-26:
Can you explain a bit on how to send key press to host?  (Initial from ESP8266. )

I am looking at source code of myTimer in user_main.c , seems to me that you were testing key press.
But the code only show build the data structure and fill the buffer, where is the action function to send?
What does CSTick( 1 );  do ?

I thought usb_send_data is used to send data, but only usb_pid_handle_in and usb_pid_handle_data use it, both looks like to receive the data from host.

cnlohr, 2016-11-06:
Sorry!  I am back after being gone for some time! Everything is actually done in user_main().  

You /really/ shouldn't worry about how it's sent :-p that's scary!  

The way it works, though is from an interrupt callback in usb.c.  With USB the host initiates all communication to the client.  All you can do is store stuff in buffers for the callback to handle schlepping data to/from the host.

cnlohr, 2022-11-09:
I really regret, I haven't built this in so long, but the idea is you need to learn how to use sockets on the ESP8266, those can just live in the main .c file and "press" the appropriate keys.  There isn't any "Send Keystrokes" in that sense.