cnlohr/espusb #16: Can we open a wiki?

zoutepopcorn, 2016-11-26:
Some people don't know howto install things. Can we open the wiki, so we can document things together?

Thanks for sharing the USP8266 :D.

cnlohr, 2016-11-27:
How does that work?  I've never done any such thing.  Can it be done with github?  Where would I set it up. The answer to this is a resounding YES!  Anything that can get more people into this is a BIG PLUS for me.  I just don't know what I'm doing!

cnlohr, 2016-11-27:
Hey! I found the button.  I just created it.  I would be up for working on it more later. Do you know if you have to be a contributor to edit it?

leopck, 2016-11-27:
I think anyone else can contribute by forking espusb then merging back. Probably there's other ways to do it?

cnlohr, 2016-11-27:
I just found a checkbox and unchecked it.  See if you can edit it now.

leopck, 2016-11-27:
Yeah, seem that we can edit now. I'm not sure if this is good since it means that anyone can change the content.

cnlohr, 2016-11-27:
Yay! I see the edit.  If someone else wouldn't mind starting to fill it out, I can edit more tomorrow.  I'm already in a documentation mode, so I'll already be writing.

Re: Who can edit:  I get two options "Collaborators ONLY" or "Anyone on Github" and I'm leaning toward the second, until it becomes an issue.

leopck, 2016-11-27:
I can help fill up on how to get started guide for users that are interested in using the espusb, make, make burn and the website.

"Anyone on Github" that sounds great, I guess you could try it out first and if there's no issue then it's fine probably? Hopefully no one tries to play pranks and delete the whole wiki that everyone would be creating...

leopck, 2016-11-27:
@cnlohr I'm just filling in basic things that I know of, if there's any mistakes just edit them away or if the format of the wiki isn't nice then just re-format it :)

zoutepopcorn, 2016-12-04:
Nice. Welk doen :)

closed 2016-12-04 by zoutepopcorn