cnlohr/espusb #17: You can get USB VID:PID for free at openmoko project.

stanson-ch, 2016-11-30:
Nice project, looks very promising.

Since you are opensource, you could easily get free VID:PID here: http://wiki.openmoko.org/wiki/USB_Product_IDs#USB_Vendor_and_Product_IDs

I got one VID:PID (1d50:6030) for one of my projects 5 years ago using one email within 1 hour. Looks like they still have free PIDs and still give out them, so just try to get one. They are nice people, e-mail them right now. :)

cnlohr, 2016-12-01:
Wouldn't this need to be per-project? Like if someone else made a project, they would use it, or are you speaking for the bootloader feature in this project, specifically?

stanson-ch, 2016-12-02:
Of course, I'm talking about bootloader only, since this is the only constant thing here. :)

cnlohr, 2016-12-02:
Hmm... you have a valid point.  Will bookmark this thread.

brainstorm, 2017-01-13:
http://pid.codes/