cnlohr/espusb #18: The default example includes a remote control web page?

T-vK, 2016-12-01:
Is all this: https://github.com/cnlohr/espusb/tree/master/web/page
part of the default example that gets flashed to the ESP when using the default makefile?   

If so, how can I access the site then? I tried http://192.168.4.1 but all I get is an empty page with the words "File not found" in to top-left corner.

cnlohr, 2016-12-01:
You need to type ```make burnweb``` so it pushes the new webpage, or you can connect to the ESP, so it's at 192.168.4.1 and type ```make netweb``` and it will push it wirelessly.

T-vK, 2016-12-01:
Hm.. `make burn` seems to be the only command that works for me. :/

I get the following error when running `make burnweb`:
```
Makefile:10: warning: overriding recipe for target 'usbburn'
esp82xx/main.mf:113: warning: ignoring old recipe for target 'usbburn'
make[1]: Entering directory '/home/esp/shared_project/web'
rm -f mfsmaker page.mpfs pushtodev execute_reflash tmp/*
gcc -DICACHE_FLASH -DWEB_PORT=80 -DCOM_PORT=7777 -DBACKEND_PORT=7878  -DSLOWTICK_MS=100 mfsmaker.c -o mfsmaker  -lusb-1.0
/usr/bin/ld: cannot find -lusb-1.0
collect2: error: ld returned 1 exit status
Makefile:17: recipe for target 'mfsmaker' failed
make[1]: *** [mfsmaker] Error 1
make[1]: Leaving directory '/home/esp/shared_project/web'
esp82xx/main.mf:98: recipe for target 'burnweb' failed
make: *** [burnweb] Error 2
```

`make netweb` is giving me:
```
Makefile:10: warning: overriding recipe for target 'usbburn'
esp82xx/main.mf:113: warning: ignoring old recipe for target 'usbburn'
make[1]: Entering directory '/home/esp/shared_project/web'
gcc -DICACHE_FLASH -DWEB_PORT=80 -DCOM_PORT=7777 -DBACKEND_PORT=7878  -DSLOWTICK_MS=100 pushtodev.c -o pushtodev  -lusb-1.0
pushtodev.c:17:31: fatal error: libusb-1.0/libusb.h: No such file or directory
compilation terminated.
Makefile:17: recipe for target 'pushtodev' failed
make[1]: *** [pushtodev] Error 1
make[1]: Leaving directory '/home/esp/shared_project/web'
esp82xx/main.mf:109: recipe for target 'netweb' failed
make: *** [netweb] Error 2
```

The ESP is definitely at 192.168.4.1, but it always responds with:
```
HTTP/1.1 404 Not Found
Connection: close

File not found.
```
when I request `GET http://192.168.4.1 HTTP/1.1`...

leopck, 2016-12-01:
@T-vK I just updated the wiki Getting Started guide with the website part. Give it a try. You can jump directly to Building website (https://github.com/cnlohr/espusb/wiki/Getting-Started-Guide#building-website) then try flashing it again.

T-vK, 2016-12-01:
I didn't even get to the point where it says "Erase: 1".

But I found out that I was just missing `libusb-1.0-0-dev`.  
A simple `apt-get install libusb-1.0-0-dev` fixed it for me.  
I'm not sure if that's in the wiki yet. 

I think I'm gonna add the dependency to my docker project, so that it can be used with espusb more conveniently.

I tested all commands except for the usbburn and usbweb is working for me. 
And there is no need to cd into the web directory. Just run `make netweb` and ignore the fact that it says "Erase: 1" at some point. It will work.

leopck, 2016-12-01:
@T-vK alright noted, i'll add the dependency for libusb and change the burning for the website. i forgotten whether i installed libusb but it does sound familiar.

T-vK, 2016-12-01:
Great. I already added instructions on how to flash the web interface using the docker method to the wiki page. I'll leave the rest for you.

leopck, 2016-12-02:
Alright it's done. Added the changes over to the wiki. If we're done then i think we can close this issue.

cnlohr, 2016-12-02:
You guys have no idea how much it excites me to see people actually making this usable!

closed 2016-12-07 by leopck