cnlohr/espusb #20: "../ld/eagle.rom.addr.v6.ld: No such file or directory"

pastapojken, 2016-12-02:
I was reflashing my esp using the new guide in the wiki and kept getting "../ld/eagle.rom.addr.v6.ld: No such file or directory" when trying to build. I edited the last line in the file with the wrong path (~/esp8266/esp-open-sdk/sdk/ld/eagle.app.v6.ld ). I simply changed it into "eagle.rom.addr.v6.ld", with that edit everything works. I'm not sure if I did anything wrong (like if my directory tree is messed up or what), but I thought I would mention it just in case.

leopck, 2016-12-02:
hi i know what's wrong. inside the user.cfg there's a commented SDK=...... please uncomment that, it's written inside the getting started guide under editing the user.cfg.

Maybe it wasn't clear enough. I'll rearrange and restructure the words there. But as for your issue, just uncomment the SDK and point the directory to your v1.5.x SDK folder.

pastapojken, 2016-12-02:
Oh wow, I read it multiple times and still missed it, sorry! 
Anyways, thanks!

closed 2016-12-02 by pastapojken