cnlohr/espusb #22: How can I add more files to the web interface?

T-vK, 2016-12-03:
I tried to add a new css and js file to the web folder, but they don't seem to get flashed properly. Changing to the existing files works just fine btw. I thought that `burnweb` should be enough, but it appears that I must be missing something.  
In my desperation I even tried to run `all`, `clean`, `purge` and `burn` ... but it didn't change anything the new files were not available on the esp.  

Any idea what I could be missing?

leopck, 2016-12-03:
Hi, maybe it's the size of your web interface? how big is your whole web interface? If it's too big, you might need to play with the user.cfg::PAGE_OFFSET.

[https://github.com/cnlohr/espusb/wiki/Getting-Started-Guide#editing-the-usercfg](https://github.com/cnlohr/espusb/wiki/Getting-Started-Guide#editing-the-usercfg)

T-vK, 2016-12-03:
But "offset" doesn't really sound like it has anything to do with the allowed size. 
I tried to double that number and now I get a part of the css file + binary gibberish where the html file should be.

leopck, 2016-12-03:
hmm, it's more like it provides more flash space for the website to load into the flash memory as seen in the code based on ESPTool flashing.

```
($(ESPTOOL_PY) $(FWBURNFLAGS)  --port $(PORT) write_flash $(PAGE_OFFSET) web/page.mpfs)||(true)
```

[https://github.com/cnlohr/esp82xx/blob/62ab44efc8d87e42db38882004503c7c8611ed12/main.mf#L99](https://github.com/cnlohr/esp82xx/blob/62ab44efc8d87e42db38882004503c7c8611ed12/main.mf#L99)

(1) I just tried a simple test for you. Written a test.js and included into the index,html. Mine seems to be working fine? How are your files like? 

(2) Do you mind posting the files and I run a test for you?

(3) And how are you checking the files, using page.mpfs?

T-vK, 2016-12-03:
I don't know what I'm doing wrong. But at the moment changes to the existing files don't even get flashed properly. Also, every time I do a `burnweb` my esp doesn't show up as an access point anymore, even after restarting it. I have to run `burn` in order to get it to show up again.  
Here are all the files that I'm trying to use: https://github.com/T-vK/espusb

I'm checking for the files by accessing them through my browser.  
E.g. `http://192.168.4.1/jquery-2.1.4.min.js.gz` does exist. But for `http://192.168.4.1/bootstrap-3.3.7.min.js.gz` I get a `File not found.`.

cnlohr, 2016-12-03:
The command would be make burnweb --- you may have run into a really really important problem! I wonder if the ESP is storing the configuration values to the place where the web stuff is!  If that's the case, I don't know how we haven't seen this before! Can you get it back to AP mode by pressing-and-holding the GPIO0 button until the LED blinks and then reboot it?

leopck, 2016-12-03:
Okay I'm starting to understd the issue here. @T-vK go to the web folder and perform a make there. You'll see the Warning on bootstrap.gz

```
Warning: Fle "bootstrap-3.3.7.min.js.gz" too long.
tmp/main.js: 6400 (6656)
Warning: Fle "reconnecting-websocket.min.js" too long.
tmp/index.html: 15360 (22016)
```

leopck, 2016-12-03:
Okay. I resolved your issue. There's two different issues on hand here. So I'll number them

1. The warning shown on ```Fle ... too long``` is due to the string of the File name is too long. Please rename the "bootstrap-3.3.7.min.js.gz" to "bs.min.js.gz and "reconnecting-websocket.min.js" to "rws.min.js" or whatever shorter names you want :) and then respectively make the changes to your index.html. This will compile them all properly without warnings.

2. The reason why you had no Access Point after flashing the website is due to the lack of space in the flash memory (I think?). So the codes are overriding each other, so I think it so happened to override the Access Point code or config files? Simple solution. Just go to ```user.cfg``` and change your value for the ```PAGE_OFFSET  = 1048576``` and that should do the trick.

@T-vK beautiful styling on that website btw. I love it.

T-vK, 2016-12-03:
@leopck Wow, thank you so much. It works! I wouldn't have figured that out in a million years...

closed 2016-12-07 by leopck