cnlohr/espusb #24: Use from esp8266/Arduino

probonopd, 2016-12-06:
Can Arduino sketches use espusb? If so, how? (I would like to avoid having to compile a toolchain and SDK, and use the ones that come with [esp8266/Arduino](https://github.com/esp8266/Arduino) instead.

Whereas #1 is about the bootloader, this ticket is about compiling e.g., a USB HID device using the Arduino IDE.

cnlohr, 2016-12-06:
As of now, I have not ported it.  It shouldn't be too difficult, but I really just cannot survive in the arduino environment.  I would appreciate it if someone were to port it, though!

T-vK, 2016-12-06:
At the moment the best I can offer you, is a docker container.  
https://github.com/cnlohr/espusb/wiki/Getting-Started-Guide#buildingflashing-under-a-docker-container
Way easier than setting up the toolchain/sdk. And no more unexpected problems.