cnlohr/espusb #32: Set static wifi station to connect

atc1441, 2017-02-24:
Hi, i dont now if this issue is correct submited but i will try :)

How can i change the wifi settings hard coded in the source code?

1. how can i set a permanent wifi station like my router and hard code the ssid and password in it? 

2. how can i rename the accespoint the esp8266 is opening but also hardcoded, where are these lines in the code. i searched everything but did not find it. 

i am comming from arduino and want to learn c programming its not that easy but defenetly better :D

atc1441, 2017-02-24:
sorry for double post, i think i found it in esp82xx/fwsrc/commonservices.c with grep but i am afraid of changing something in it.
will try some things :)

cnlohr, 2017-02-25:
You can actually use the normal functions available in esp nonos sdk.

atc1441, 2017-02-25:
After some reading and google searching i was able to get it to work. 
Here i got example codes: [https://espressif.com/en/support/explore/sample-codes](https://espressif.com/en/support/explore/sample-codes)

Just add the needed code in the file/user/user_main.c under the segment: user_init(void)

i think you can find so less information about the sdk and esp8266, all is just in arduino.
for example i want to build something with the i2C and the sdk for the esp but its so hard to find a working example

cnlohr, 2017-02-25:
Are you an I2C master?  If so checkout what I have in wiflier. https://github.com/cnlohr/wiflier/tree/master/espfirmware/user

atc1441, 2017-02-25:
thank you for that.
i am at the begining of understanding C and how to comdine it will try it :)

stb68, 2020-04-13:
Hi @atc1441, the link in your post above to example code doesn't work now, do you have an example of what to add to make the ESP connect to an existing network on startup?
Thanks

atc1441, 2020-04-14:
Hey. Sorry i need to search for it. 

But look into the user init function maybe that helps you

stb68, 2020-04-14:
Thanks, when I looked in there I couldn't see anything obvious to change so I'm thinking it'a all new bits to add in there.

stb68, 2020-04-15:
If anybody else is trying this then I've found the code examples (previously at https://espressif.com/en/support/explore/sample-codes) here - https://web.archive.org/web/20170202091829/https://espressif.com/en/support/explore/sample-codes

Not sure what to do with them as I've done very little C programming spread very sparsly over a very long time and this is my first attempt at ESP but I'll give it a go.

timg11, 2022-02-11:
If anyone is looking to make this work, I posted a modified version of user_main.c that initializes to Station mode (using a hard-coded SSID and WPA) to the [ESP8266 forum](https://www.esp8266.com/viewtopic.php?f=156&t=20439&p=93628#p93628).