cnlohr/espusb #33: Pre-Compiled?

detectingdragon, 2017-07-05:
I cant seem to compile it! Is it possible to get a pre-compiled copy for esp8266 12E?

T-vK, 2017-07-05:
Have you checked out the wiki? https://github.com/cnlohr/espusb/wiki/Getting-Started-Guide   
If you tell us where exactly you are stuck in the compiling process, we might be able to help you.

detectingdragon, 2017-07-05:
I get this error:
[INFO ]  Extracting and patching toolchain components
[14:51] | /home/pi/esp8266/esp-open-sdk/crosstool-NG/lib/crosstool-ng-1.22.0-60-g37b07f6/scripts/functions: line 198: printf: write error: No space left on device
/home/pi/esp8266/esp-open-sdk/crosstool-NG/lib/crosstool-ng-1.22.0-60-g37b07f6/scripts/functions: line 198: printf: write error: No space left on device
[ERROR]  
[14:52] / /home/pi/esp8266/esp-open-sdk/crosstool-NG/lib/crosstool-ng-1.22.0-60-g37b07f6/scripts/functions: line 198: printf: write error: No space left on device
[ERROR]  >>
[14:52] / /home/pi/esp8266/esp-open-sdk/crosstool-NG/lib/crosstool-ng-1.22.0-60-g37b07f6/scripts/functions: line 198: printf: write error: No space left on device
[ERROR]  >>  Build failed in step 'Extracting and patching toolchain components'
[14:52] / /home/pi/esp8266/esp-open-sdk/crosstool-NG/lib/crosstool-ng-1.22.0-60-g37b07f6/scripts/functions: line 198: printf: write error: No space left on device
[ERROR]  >>        called in step '(top-level)'
[14:52] / /home/pi/esp8266/esp-open-sdk/crosstool-NG/lib/crosstool-ng-1.22.0-60-g37b07f6/scripts/functions: line 198: printf: write error: No space left on device
[ERROR]  >>
[14:52] / /home/pi/esp8266/esp-open-sdk/crosstool-NG/lib/crosstool-ng-1.22.0-60-g37b07f6/scripts/functions: line 198: printf: write error: No space left on device
[ERROR]  >>  Error happened in: CT_DoLog[scripts/functions@198]
[14:52] / /home/pi/esp8266/esp-open-sdk/crosstool-NG/lib/crosstool-ng-1.22.0-60-g37b07f6/scripts/functions: line 198: printf: write error: No space left on device
[ERROR]  >>        called from: CT_DoExecLog[scripts/functions@257]
[14:52] / /home/pi/esp8266/esp-open-sdk/crosstool-NG/lib/crosstool-ng-1.22.0-60-g37b07f6/scripts/functions: line 198: printf: write error: No space left on device
[ERROR]  >>        called from: CT_Extract[scripts/functions@984]
[14:52] / /home/pi/esp8266/esp-open-sdk/crosstool-NG/lib/crosstool-ng-1.22.0-60-g37b07f6/scripts/functions: line 198: printf: write error: No space left on device
[ERROR]  >>        called from: do_gcc_extract[scripts/build/cc/100-gcc.sh@57]
[14:52] / /home/pi/esp8266/esp-open-sdk/crosstool-NG/lib/crosstool-ng-1.22.0-60-g37b07f6/scripts/functions: line 198: printf: write error: No space left on device
[ERROR]  >>        called from: do_cc_extract[scripts/build/cc.sh@28]
[14:52] / /home/pi/esp8266/esp-open-sdk/crosstool-NG/lib/crosstool-ng-1.22.0-60-g37b07f6/scripts/functions: line 198: printf: write error: No space left on device
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@616]
[14:52] / /home/pi/esp8266/esp-open-sdk/crosstool-NG/lib/crosstool-ng-1.22.0-60-g37b07f6/scripts/functions: line 198: printf: write error: No space left on device
[ERROR]  >>
[14:52] / /home/pi/esp8266/esp-open-sdk/crosstool-NG/lib/crosstool-ng-1.22.0-60-g37b07f6/scripts/functions: line 198: printf: write error: No space left on device
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[14:52] / /home/pi/esp8266/esp-open-sdk/crosstool-NG/lib/crosstool-ng-1.22.0-60-g37b07f6/scripts/functions: line 198: printf: write error: No space left on device
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[14:52] / /home/pi/esp8266/esp-open-sdk/crosstool-NG/lib/crosstool-ng-1.22.0-60-g37b07f6/scripts/functions: line 198: printf: write error: No space left on device
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6/B - Known issues.txt'
[14:52] / /home/pi/esp8266/esp-open-sdk/crosstool-NG/lib/crosstool-ng-1.22.0-60-g37b07f6/scripts/functions: line 198: printf: write error: No space left on device
[ERROR]  
[14:52] / /home/pi/esp8266/esp-open-sdk/crosstool-NG/lib/crosstool-ng-1.22.0-60-g37b07f6/scripts/functions: line 198: printf: write error: No space left on device
[ERROR]  (elapsed: 14:53.49)
[14:53] / ct-ng:152: recipe for target 'build' failed
make[2]: *** [build] Error 1
make[2]: Leaving directory '/home/pi/esp8266/esp-open-sdk/crosstool-NG'
../Makefile:132: recipe for target '_toolchain' failed
make[1]: *** [_toolchain] Error 2
make[1]: Leaving directory '/home/pi/esp8266/esp-open-sdk/crosstool-NG'
Makefile:128: recipe for target '/home/pi/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc' failed
make: *** [/home/pi/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2

T-vK, 2017-07-06:
How much disk space do you have left on your device?

detectingdragon, 2017-07-07:
I tried using a different Linux system. Here is what i got when running "make"

[INFO ]  Installing GMP for host
[ERROR]    make[5]: *** [div_qr_1n_pi1.lo] Error 1
[ERROR]    make[4]: *** [all-recursive] Error 1
[ERROR]    make[3]: *** [all] Error 2
[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing GMP for host'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: do_gmp_backend[scripts/build/companion_libs/100-gmp.sh@106]
[ERROR]  >>        called from: do_gmp_for_host[scripts/build/companion_libs/100-gmp.sh@62]
[ERROR]  >>        called from: do_companion_libs_for_host[scripts/build/companion_libs.sh@36]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@646]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 22:06.22)
[22:08] / ct-ng:152: recipe for target 'build' failed
make[2]: *** [build] Error 2
make[2]: Leaving directory '/home/detectingdragon77/esp8266/firmware/esp-open-sdk/crosstool-NG'
../Makefile:132: recipe for target '_toolchain' failed
make[1]: *** [_toolchain] Error 2
make[1]: Leaving directory '/home/detectingdragon77/esp8266/firmware/esp-open-sdk/crosstool-NG'
Makefile:128: recipe for target '/home/detectingdragon77/esp8266/firmware/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc' failed
make: *** [/home/detectingdragon77/esp8266/firmware/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2

My build log is attached.

T-vK, 2017-07-07:
Can you please tell us what you have done so far? Have you followed the steps in the wiki?  
Also, what Linux distro are you using?   
Is your kernel version >=3.10 ? If so, then you could simply use the docker image that I created. That would make thinks much easier.

detectingdragon, 2017-07-07:
Ok, I am running Armbian on an Orange Pi One, my arch is armhf: 
detectingdragon77@orangepione:~$ uname -a
Linux orangepione 3.4.113-sun8i #20 SMP PREEMPT Sun Jun 18 10:55:08 CEST 2017 armv7l armv7l armv7l GNU/Linux

I attached a text file of all my terminal commands.

detectingdragon, 2017-07-07:
Nevermind here:

 detectingdragon77@orangepione:~$ uname -a
Linux orangepione 3.4.113-sun8i #20 SMP PREEMPT Sun Jun 18 10:55:08 CEST 2017 armv7l armv7l armv7l GNU/Linux
detectingdragon77@orangepione:~$ sudo apt-get install make unrar autoconf automake libtool gcc g++ gperf \
>     flex bison texinfo gawk ncurses-dev libexpat-dev python-dev python python-serial \
>     sed git unzip bash help2man wget bzip2 libtool-bin
[sudo] password for detectingdragon77: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libncurses5-dev' instead of 'ncurses-dev'
Note, selecting 'libexpat1-dev' instead of 'libexpat-dev'
autoconf is already the newest version (2.69-9).
automake is already the newest version (1:1.15-4ubuntu1).
bison is already the newest version (2:3.0.4.dfsg-1).
bzip2 is already the newest version (1.0.6-8).
flex is already the newest version (2.6.0-11).
g++ is already the newest version (4:5.3.1-1ubuntu1).
gawk is already the newest version (1:4.1.3+dfsg-0.1).
gcc is already the newest version (4:5.3.1-1ubuntu1).
libncurses5-dev is already the newest version (6.0+20160213-1ubuntu1).
libtool is already the newest version (2.4.6-0.1).
libtool-bin is already the newest version (2.4.6-0.1).
make is already the newest version (4.1-6).
python is already the newest version (2.7.11-1).
python-dev is already the newest version (2.7.11-1).
python-serial is already the newest version (3.0.1-1).
sed is already the newest version (4.2.2-7).
unzip is already the newest version (6.0-20ubuntu1).
gperf is already the newest version (3.0.4-2).
help2man is already the newest version (1.47.3).
texinfo is already the newest version (6.1.0.dfsg.1-5).
unrar is already the newest version (1:5.3.2-1).
bash is already the newest version (4.3-14ubuntu1.2).
git is already the newest version (1:2.7.4-0ubuntu1.1).
libexpat1-dev is already the newest version (2.1.0-7ubuntu0.16.04.2).
wget is already the newest version (1.17.1-1ubuntu1.2).
0 upgraded, 0 newly installed, 0 to remove and 6 not upgraded.
detectingdragon77@orangepione:~$ clear
[3;J
detectingdragon77@orangepione:~$ cd esp8266
detectingdragon77@orangepione:~/esp8266$ cd
detectingdragon77@orangepione:~$ clear
[3;J

detectingdragon77@orangepione:~$ cd esp8266
detectingdragon77@orangepione:~/esp8266$ git clone --recursive https://github.com/pfalcon/esp-open-sdk.git
Cloning into 'esp-open-sdk'...
remote: Counting objects: 530, done.
remote: Total 530 (delta 0), reused 0 (delta 0), pack-reused 530
Receiving objects: 100% (530/530), 329.81 KiB | 22.00 KiB/s, done.
Resolving deltas: 100% (306/306), done.
Checking connectivity... done.
Submodule 'crosstool-NG' (https://github.com/jcmvbkbc/crosstool-NG) registered for path 'crosstool-NG'
Submodule 'esp-open-lwip' (https://github.com/pfalcon/esp-open-lwip) registered for path 'esp-open-lwip'
Submodule 'esptool' (https://github.com/themadinventor/esptool) registered for path 'esptool'
Submodule 'lx106-hal' (https://github.com/tommie/lx106-hal) registered for path 'lx106-hal'
Cloning into 'crosstool-NG'...
remote: Counting objects: 33801, done.
Receiving objects:  10% (3681/33801), 1.89 MiReceiving objects:  11% (3719/33801), 1.93 MiReceiving objects:  11% (3758/33801), 1.96 MiReceiving objects:  11% (3911/33801), 2.04 MiReceiving objects:  11% (4040/33801), 2.07 MiReceiving objects:  12% (4057/33801), 2.11 MiReceiving objects:  12% (4069/33801), 2.14 MiReceiving objects:  12% (4091/33801), 2.23 MiReceiving objects:  12% (4091/33801), 2.32 MiReceiving objects:  12% (4091/33801), 2.41 MiReceiving objects:  12% (4091/33801), 2.50 MiReceiving objects:  12% (4182/33801), 2.69 MiReceiving objects:  12% (4183/33801), 2.75 MiReceiving objects:  12% (4291/33801), 2.90 MiReceiving objects:  12% (4385/33801), 2.94 MiReceiving objects:  13% (4395/33801), 3.07 MiReceiving objects:  13% (4614/33801), 3.09 MiReceiving objects:  14% (4733/33801), 3.39 MiReceiving objects:  14% (4751/33801), 3.64 MiReceiving objects:  14% (4751/33801), 3.87 MiReceiving objects:  14% (4751/33801), 4.04 MiReceiving objects:  14% (4751/33801), 4.23 MiReceiving objects:  26% (8937/33801), 8.42 MiReceiving objects:  27% (9127/33801), 8.53 MiReceiving objects:  27% (9456/33801), 8.62 MiReceiving objects:  28% (9465/33801), 8.62 MiReceiving objects:  29% (9803/33801), 8.70 MiReceiving objects:  29% (10129/33801), 8.79 MReceiving objects:  30% (10141/33801), 8.79 MReceiving objects:  31% (10479/33801), 8.87 MReceiving objects:  31% (10733/33801), 8.96 MReceiving objects:  32% (10817/33801), 8.96 MReceiving objects:  33% (11155/33801), 9.04 MReceiving objects:  33% (11245/33801), 9.13 Mremote: Total 33801 (delta 0), reused 0 (delta 0), pack-reused 33800
Receiving objects: 100% (33801/33801), 17.04 MiB | 169.00 KiB/s, done.
Resolving deltas: 100% (19981/19981), done.
Checking connectivity... done.
Submodule path 'crosstool-NG': checked out '37b07f6fbea2e5d23434f7e91614528f839db056'
Cloning into 'esp-open-lwip'...
remote: Counting objects: 236, done.
remote: Total 236 (delta 0), reused 0 (delta 0), pack-reused 236
Receiving objects: 100% (236/236), 495.14 KiB | 393.00 KiB/s, done.
Resolving deltas: 100% (75/75), done.
[... 247 lines cut ...]
[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing GMP for host'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[... cut ...]
make[2]: Leaving directory '/home/detectingdragon77/esp8266/esp-open-sdk/crosstool-NG'
../Makefile:132: recipe for target '_toolchain' failed
make[1]: *** [_toolchain] Error 2
make[1]: Leaving directory '/home/detectingdragon77/esp8266/esp-open-sdk/crosstool-NG'
Makefile:128: recipe for target '/home/detectingdragon77/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc' failed
make: *** [/home/detectingdragon77/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2
detectingdragon77@orangepione:~/esp8266/esp-open-sdk$

cnlohr, 2017-09-09:
Did you ever get any further on this?  I don't have any armbian devices ATM so it's a bit tricky for me.

zoobab, 2017-12-24:
I will try to rework my Dockerfile for esp-usb and push it this week.

sinaj2889, 2022-06-17:
Can you upload a .ino or a .bin to the repo?