cnlohr/espusb #34: WiFi Password

Josh-Hogan, 2017-08-21:
Hi there does anyone know if it would be possible to set a WiFi password for the hosted ap I haven't been able to find it anywhere or even just hide the ssid would do?
Thanks

cnlohr, 2017-09-09:
You should be able to type the password into the box in the web gui and it should JustWork™ But, if it doesn't report back.  It could be an issue with esp82xx.

joBr99, 2018-04-01:
I can confirm that setting a wifi password in ap mode does not work.

[image]

cnlohr, 2018-04-01:
Bummer. You're right.  I don't know why I diasbled it.  It's been disabled both in the web GUI (Easy to override) and in commonservices.c:442.  If someone could take a gander and try recompiling with both enabled and seeing if it works, it could just need to be "turned on"

Blast12345, 2018-09-16:
@cnlohr 

I modified the web GUI to allow that field to be enabled.
I attempted to modify commonservices.c, but didn't have much luck.

Unmodified:
```
    				#if 0 //We don't support encryption.
					config.ssid[c1l] = 0;  config.password[c2l] = 0;   config.authmode = 0;
					if( encr ) {
						int k;
						for( k = 0; enctypes[k]; k++ )
							if( strcmp( encr, enctypes[k] ) == 0 )
								config.authmode = k;
					}
    				#endif
```

After modification:
```
					//c1l and c2l are undefined
					//config.ssid[c1l] = 0;  config.password[c2l] = 0;   config.authmode = 0;
					if( encr ) {
						int k;
						for( k = 0; enctypes[k]; k++ )
							if( strcmp( encr, enctypes[k] ) == 0 )
								config.authmode = k;
					}
```

While this did compile, my ESP8266 did not connect to the wifi.