cnlohr/espusb #36: USB HID MIDI Device?

Boscop, 2017-09-13:
Is it possible to make the ESP8266 appear as a HID MIDI device over USB with this library?
If so, what are the necessary steps?

leopck, 2017-09-13:
Not that I know of currently, I think it's not impossible to get this done. You'll just need the USB descriptor from the HID MIDI device and compile it together with espusb. This should make espusb look like a HID MIDI device.

cnlohr, 2017-09-14:
AFAIK, modern OSes for some reason have started banning low-speed devices from being more sophisticated, to be in-line with the USB standard.  I REALLY WISH THEY DIDN'T DO THIS! But it does mean things like MIDI, CDC, RMDIS, mass storage, etc. are all prohibited on low-speed systems :(.

There've been a lot of people complaining and for some reason nothing's been done to allow it.

Boscop, 2017-09-15:
But it's possible with the Attiny85 / Digispark and Arduino (with HIDUINO):
https://github.com/heartscrytech/DigisparkMIDI
https://github.com/ddiakopoulos/hiduino
https://github.com/NicoHood/HID

And those are also low-speed systems, right?

cnlohr, 2017-09-15:
I tried this about a year ago, and got an error message saying it "didn't have enough bandwidth."

Would you be willing to try on your system?  What is your system?  Windows? Linux?

Boscop, 2017-09-15:
Windows 8.1. Yes, I'm willing to try?
What exactly should I do? :)

cnlohr, 2017-09-16:
get one of the existing low-speed devices, get them to enumerate and work with Windows!