cnlohr/espusb #39: Wiki - Getting Started Guide

wlleong, 2018-01-18:
I'm new to this, not sure if this is the place to post it, feel free to move it somewhere else.

On the Getting Started Guide, under "Building/flashing under a Docker container", to build the docker image, the final step should be `sudo docker build -t tavk/esp-sdk:0.1.0 .` with a space followed by a full stop at the end.

cnlohr, 2018-01-19:
I've never used docker so I don't know where that would fit in context.  Mind giving me some context?

wlleong, 2018-01-23:
Yes certainly. So in the Wiki of this GitHub repo, under Getting Started Guide (https://github.com/cnlohr/espusb/wiki/Getting-Started-Guide) two methods are presented to build and flash espusb. The first method involves downloading and building the Espressif SDK manually. The second method downloads and builds a docker image with the SDK configured.

As mentioned in the original post, to build the docker image, the final step says `sudo docker build -t tavk/esp-sdk:0.1.0` but it should instead be `sudo docker build -t tavk/esp-sdk:0.1.0 .` essentially adding a space followed by a fullstop at the end for it to work.