cnlohr/espusb #40: Nodemcu with espusb

openscoreboard, 2018-02-17:
I have a nodemcu
I can flash espusb inside
I can connect my smartphone to 192.168.4.1
But i lost connection if push web buttons...

What schematic need for connect a nodemcu and work with espusb in mode hid?

Thank!!!

cnlohr, 2018-02-17:
Please see schematic info on this page: https://github.com/cnlohr/espusb

Charles

openscoreboard, 2018-02-21:
I try flash to 0x100000 but the connection from smartphone lost again....
I see schematic too, but my nodemcu is a little diferent...

Is It ok my connection?

[image]

Thanks...

cnlohr, 2018-02-22:
What resistor are you using?  1.5k is nominal.   Connection should be ok.

Do you have access to a Linux computer, using ```dmesg``` it can make it much easier to understand what's going on.

openscoreboard, 2018-03-03:
I have a resistor with 1,2K  

And this is the output dmesg

> [sáb mar  3 13:23:35 2018] ch341 2-2:1.0: device disconnected
[sáb mar  3 13:23:40 2018] usb 2-2: new full-speed USB device number 5 using uhci_hcd
[sáb mar  3 13:23:40 2018] usb 2-2: New USB device found, idVendor=1a86, idProduct=7523
[sáb mar  3 13:23:40 2018] usb 2-2: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[sáb mar  3 13:23:40 2018] usb 2-2: Product: USB2.0-Serial
[sáb mar  3 13:23:40 2018] ch341 2-2:1.0: ch341-uart converter detected
[sáb mar  3 13:23:40 2018] usb 2-2: ch341-uart converter now attached to ttyUSB0

> 
My steps

- Connect NODEMCU by USB to PC (linux)
- run commands make burn and make burnweb (my nodemcu blink the  led)
-Disconnect USB
-Connect GPIO4+resistor+3v
-Reconnect USB to PC
-I think that no happend nothing 

Thanks...

cnlohr, 2018-03-03:
uuhhhh ... can you take a picture of your setup?  I don't see where you're hooking the second USB in.  Also, you should never need to unplug the NodeMCU usb. Just plug both USB plugs in.

openscoreboard, 2018-03-03:
I have only one USB port connect to my PC.
This is my connection
[image]

openscoreboard, 2018-03-03:
oooohhhh!!!
I think I understood it now.
It is necessary to connect a second USB to the GPIO5 (D +) and GPIO4 (D-) pin.
This second USB is the works as HID
Is correct?

cnlohr, 2018-03-03:
YES

openscoreboard, 2018-03-04:
Now this is my setup

[image]

One USB for flash and other for HID

This is my dmesg now...

> [  263.532501] ch341 3-1:1.0: device disconnected
[  270.544035] usb 3-1: new full-speed USB device number 4 using uhci_hcd
[  270.719080] usb 3-1: New USB device found, idVendor=1a86, idProduct=7523
[  270.719085] usb 3-1: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[  270.719089] usb 3-1: Product: USB2.0-Serial
[  270.721268] ch341 3-1:1.0: ch341-uart converter detected
[  270.728196] usb 3-1: ch341-uart converter now attached to ttyUSB0
[  280.808041] usb 5-2: new full-speed USB device number 14 using uhci_hcd
[  281.148045] usb 5-2: device descriptor read/64, error -71
[  281.384144] usb 5-2: device descriptor read/64, error -71
[  281.612035] usb 5-2: new full-speed USB device number 15 using uhci_hcd
[  281.740030] usb 5-2: device descriptor read/64, error -71
[  281.976033] usb 5-2: device descriptor read/64, error -71
[  282.204032] usb 5-2: new full-speed USB device number 16 using uhci_hcd
[  282.620032] usb 5-2: device not accepting address 16, error -71
[  282.740030] usb 5-2: new full-speed USB device number 17 using uhci_hcd
[  283.156030] usb 5-2: device not accepting address 17, error -71
[  283.156061] usb usb5-port2: unable to enumerate USB device

>

cnlohr, 2018-03-04:
It is likely you've either flipped the D+/D- pins or some issue with the pullup resistor being large enough, etc.

openscoreboard, 2018-03-31:
The last week i tried connect one wemos d1 and dont work for me.

I think that my firmware isnt ok.
Is there a link to donwload the firmware pre-compiled?

Thanks

joBr99, 2018-03-31:
In this repository is a pre compiled firmware. You can flash it with the esptool and the following command:
esptool.py -b 115200 --port /dev/ttyUSB0 write_flash 65536 web/page.mpfs

As I have already said it does not work for me, I will attach my compiled firmware too.

[page.mpfs.zip](https://github.com/cnlohr/espusb/files/1865763/page.mpfs.zip)

cnlohr, 2018-04-01:
The page.mpfs is only the webpage... you'll need both the 0x0000 and 0x4000 binaries.  I don't have them handy, sadly.

joBr99, 2018-04-01:
Thanks, I overlooked that ... now it is working.

cnlohr, 2018-04-01:
@joBr99 Would you mind uploading your binaries?

cnlohr, 2018-04-01:
I am posting this thread to the main readme.

closed 2018-04-01 by cnlohr

ntinos97, 2020-05-12:
> esptool.py -b 115200 --port /dev/ttyUSB0 write_flash -fm dio 0x00000 image.elf-0x00000.bin 0x40000 image.elf-0x40000.bin
> esptool.py -b 115200 --port /dev/ttyUSB0 write_flash 65536 web/page.mpfs
> 
> [espusb.zip](https://github.com/cnlohr/espusb/files/1866448/espusb.zip)

love you...i had all day until to find this solution... thank you so much

NicoAlmazan, 2022-01-25:
Hello guys.. I think I f*** it up. Today I boght this Module I share with you in the picture. 
[image]

I tryed to make it work but I couldn´t. I even flashed it on the memory addresses you shared. image.elf-0x00000.bin 0x40000 image.elf-0x40000.bin and web/page.mpfs to 0x100000
It kindda work.. but it is not the interface I am used to see on videos and it does not even send commands to the module.
This is what everybody sees:

[image]

This is what I see:

[image]

Can you guys please help me? 
Thank you