cnlohr/espusb #46: wemos d1 issues

toremaannn, 2018-07-03:
Hi I am having trouble with getting espusb to work. I have managed to get the software to work. (I think) The website is working an "Messages: is increasing when I reload the side.  But when I am unable to send key strokes  to the computer

When I connect it to a ubuntu machine I am seeing this in the dmesg 

`[ 8670.728659] usb 2-2: device descriptor read/64, error -71
[ 8670.836665] usb usb2-port2: attempt power cycle
[ 8671.488592] usb 2-2: new full-speed USB device number 73 using xhci_hcd
[ 8671.488807] usb 2-2: Device not responding to setup address.
[ 8671.696671] usb 2-2: Device not responding to setup address.
[ 8671.904508] usb 2-2: device not accepting address 73, error -71
[ 8672.032503] usb 2-2: new full-speed USB device number 74 using xhci_hcd
[ 8672.032658] usb 2-2: Device not responding to setup address.
[ 8672.240634] usb 2-2: Device not responding to setup address.
[ 8672.448482] usb 2-2: device not accepting address 74, error -71
[ 8672.448550] usb usb2-port2: unable to enumerate USB device`

When I plug it in a windows machine it is recognized as a unknown device

Hardware is not my strongest side, but I really need this to work 

Her is some pictures of how I have connected the HW
https://photos.app.goo.gl/QQ4c7XgAZnopFRRw

Can someone point me in the right directions.

enricolam, 2018-07-03:
I got 404 for the link u provide

toremaannn, 2018-07-03:
Fixed the link

cnlohr, 2018-07-03:
It "looks" like a wiring problem ,like d+/d- are flipped or you have a resistor on the wrong leg.

cnlohr, 2018-07-03:
Whoops responded before looking at the pics.  Looking now.

cnlohr, 2018-07-03:
I am unfamiliar with the pinout of the Wemos.  can you draw me a rough schematic (even if it's just ms paint)?

toremaannn, 2018-07-03:
I had the resistor on the wrong leg. Thank you very much!!