cnlohr/espusb #47: Windows - Unknown USB Device (Device Descriptor Request Failed

toremaannn, 2018-07-04:
Hi

Do I need any driver on a windows 2016 server? working fine on a ubuntu, macOS and windows 7

In device manager I am seeing this: Unknown USB Device (Device Descriptor Request Failed

cnlohr, 2018-07-04:
I don't believe so.  It should work just fine, considering it follows the normal HID classes.  The fact that it says that is a little concerning like maybe it's operating somewhat differently than Windows is expecting.  You will need to run USBPcap to capture the transaction and see what exactly windows is asking and what the device is saying.

blueman007, 2018-10-30:
For the poor soul who will inevitably read this post one day: if you have a Device Descriptor error on Windows, it's very likely that your D+ and D- cables are swapped. Don't ask me how I know.

You're welcome.

myfingerhurt, 2021-03-24:
@blueman007 
Thank you very much, you saved my day!
And for the recorder, the D4 and D5 on NodeMCU are not GPIO5 and GPIO4.

I have also found out that ESP-Mx DevKit aka 2AL3B-ESP-M or ESP-M2 is not working with ESPUSB.
The Wi-Fi web-server also dose not start, terminal with 115200 talks gibberish.

[image]