cnlohr/espusb #51: Manufacturing an EspUSB

peterburk, 2019-08-05:
I've been wanting to buy an EspUSB since I first heard about it in 2016. Unfortunately the design, though free and open-source, is incomprehensible to any contract manufacturers. I only have access to a large chisel soldering iron, and rarely have a stable address for shipping components to me. I need someone else to build it, and I will pay. I want to be able to type foreign languages (e.g. Chinese) directly into my laptop just by changing the virtual keys on my phone. 

A year ago I tried to ask PCBWay to build it. They said they want a BOM file. I knew that this could be deduced from the schematic, but I don't trust my guesswork. I asked on the [forum](https://www.esp8266.com/viewtopic.php?f=155&t=19928), without any success. I asked in the Hackaday chatroom, without success. Finally I was visiting a colleague, and they helped me reverse-engineer the BOM from the sch file. PCBWay rejected it again, wanting me to provide Digikey part numbers. A couple of hours of searching and guessing later, I hope I guessed those right. If anybody needs the BOM, I'm attaching that. It really should be included in the design folder. At this point, PCBWay took the order, and let me pay. 

[358_BOM.xlsx](https://github.com/cnlohr/espusb/files/3467454/358_BOM.xlsx)

Now they're asking for a "pick & place file". My colleague said to just send PCBWay a photo, but they won't accept it, they want a "centroid file". I don't know how to make one of those. I've already guessed a lot, and I think there's only a 50% chance that the finished product will work, because I've had to make up random numbers for half the design process. I guess I'll lose the money I've already paid to PCBWay. 

Please can someone tell me how to get the EspUSB Tiny manufactured?

ztipnis, 2019-08-07:
Try MacroFab (https://macrofab.com)
They accept brd files from Eagle, and all you then have to do is use their tool to confirm that they guessed the correct part for each footprint. It’s not cheap by any means but has served me well once or twice

peterburk, 2019-08-07:
Thanks ztipnis! I tried uploading the design to MacroFab, and it's certainly easier to use than PCBWay. Unfortunately I can't find an antenna for the BOM in their parts list. 
Thank you also for suggesting Kicad, I tried exporting a POS file from there and sent that to PCBWay to see if they'll understand it instead of a centroid file. We'll see... I really hope this works after all the guesswork of choosing parts and suppliers.

ztipnis, 2019-08-07:
Usually antennas are SMA or routed in the PCB. When I’m at my computer I’ll take a look and see what antenna solution they have devised. 

Otherwise, if you’ve found a supplier for the antenna already, you can have them shipped to macrofab who can then add them to the to your design. It would take longer but is an option. You also have the option of modifying the pcb in kicad to use a part that macrofab provides, or an SMA or onboard antenna.

cnlohr, 2019-12-20:
Honestly, any chip antenna that's for 2.4 GHz and physically fits there is fine.