cnlohr/espusb #56: Any plans to support esp32?

victornor, 2020-06-08:
Excuse my lack of knowledge, if this is not even possible.

TheEngineersCorner, 2020-08-04:
have you any update about this?
victor did you  explope it working on esp32.

h0oni, 2020-12-29:
Please, someone, help the noobies like me and create support for esp32.

cnlohr, 2021-01-11:
This is very difficult to do.  And, I have moved onto other Espressif chips.  I would encourage support from anyone who does it but do not plan to contribute, myself.

lllaaa, 2022-02-09:
esp32 uses Xtensa LX6 core which is a multi-stage pipelined core. It's very difficult to control the instruction cycles.

> Excuse my lack of knowledge, if this is not even possible.

tavonkelly, 2023-03-05:
@lllaaa doesn't the ESP8266 also use Xtensa LX6?

cnlohr, 2023-03-15:
yes