cnlohr/espusb #57: esp8266/esp-open-sdk/sdk/lib/libmain.a Not found

SinanAkkoyun, 2020-09-18:
Hi, I can not make your project...

```
make ESP_ROOT=~/esp8266/esp-open-sdk burn
Makefile:10: warning: overriding recipe for target 'usbburn'
esp82xx/main.mf:113: warning: ignoring old recipe for target 'usbburn'
/home/debian/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -mlongcalls -Os -I/home/debian/esp8266/esp-open-sdk/sdk/include -Iesp82xx/include -I. -Iesp82xx/fwsrc -Iuser -DICACHE_FLASH -DWEB_PORT=80 -DCOM_PORT=7777 -DBACKEND_PORT=7878  -DSLOWTICK_MS=100 -DVERSSTR='"Version: 0a60b-dev - Build Sat, 19 Sep 2020 00:37:33 +0200 with -DICACHE_FLASH -DWEB_PORT=80 -DCOM_PORT=7777 -DBACKEND_PORT=7878  -DSLOWTICK_MS=100"'  esp82xx/fwsrc/uart.c esp82xx/fwsrc/esp82xxutil.c esp82xx/fwsrc/flash_rewriter.c esp82xx/fwsrc/http.c esp82xx/fwsrc/commonservices.c esp82xx/fwsrc/http_custom.c esp82xx/fwsrc/mdns.c esp82xx/fwsrc/mfs.c user/custom_commands.c user/user_main.c user/usb_asm_1bit.S user/usb.c -flto -Wl,--relax -Wl,--gc-sections -nostdlib -L/home/debian/esp8266/esp-open-sdk/sdk/lib -L/home/debian/esp8266/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/libgcc.a /home/debian/esp8266/esp-open-sdk/sdk/lib/libmain.a /home/debian/esp8266/esp-open-sdk/sdk/lib/liblwip.a /home/debian/esp8266/esp-open-sdk/sdk/lib/libssl.a /home/debian/esp8266/esp-open-sdk/sdk/lib/libupgrade.a /home/debian/esp8266/esp-open-sdk/sdk/lib/libnet80211.a /home/debian/esp8266/esp-open-sdk/sdk/lib/libwpa.a /home/debian/esp8266/esp-open-sdk/sdk/lib/libphy.a /home/debian/esp8266/esp-open-sdk/sdk/lib/liblwip.a /home/debian/esp8266/esp-open-sdk/sdk/lib/libcrypto.a /home/debian/esp8266/esp-open-sdk/sdk/lib/libpp.a /home/debian/esp8266/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/libgcc.a -T /home/debian/esp8266/esp-open-sdk/sdk/ld/eagle.app.v6.ld -B/home/debian/esp8266/esp-open-sdk/sdk/lib  -o image.elf
xtensa-lx106-elf-gcc: error: /home/debian/esp8266/esp-open-sdk/sdk/lib/libmain.a: No such file or directory
xtensa-lx106-elf-gcc: error: /home/debian/esp8266/esp-open-sdk/sdk/lib/liblwip.a: No such file or directory
xtensa-lx106-elf-gcc: error: /home/debian/esp8266/esp-open-sdk/sdk/lib/libssl.a: No such file or directory
xtensa-lx106-elf-gcc: error: /home/debian/esp8266/esp-open-sdk/sdk/lib/libupgrade.a: No such file or directory
xtensa-lx106-elf-gcc: error: /home/debian/esp8266/esp-open-sdk/sdk/lib/libnet80211.a: No such file or directory
xtensa-lx106-elf-gcc: error: /home/debian/esp8266/esp-open-sdk/sdk/lib/libwpa.a: No such file or directory
xtensa-lx106-elf-gcc: error: /home/debian/esp8266/esp-open-sdk/sdk/lib/libphy.a: No such file or directory
xtensa-lx106-elf-gcc: error: /home/debian/esp8266/esp-open-sdk/sdk/lib/liblwip.a: No such file or directory
xtensa-lx106-elf-gcc: error: /home/debian/esp8266/esp-open-sdk/sdk/lib/libcrypto.a: No such file or directory
xtensa-lx106-elf-gcc: error: /home/debian/esp8266/esp-open-sdk/sdk/lib/libpp.a: No such file or directory
```

Please help

SinanAkkoyun, 2020-09-20:
Hello, btw I was following your whole project instructions, but could not get it to work. I use linux wsl, but nontheless, compiling the sdk with crosstool ng and stuff worked out flawlessly...

myfingerhurt, 2021-03-24:
1. Download ESP8266_NONOS_SDK_V1.5.4_16_05_20.zip
`wget -O ESP8266_NONOS_SDK_V1.5.4_16_05_20.zip https://bbs.espressif.com/download/file.php?id=1469&sid=3e67f0c75557051412485c5d4e578de7`

2. unpack zip
`unzip ESP8266_NONOS_SDK_V1.5.4_16_05_20.zip -d ~/esp8266/SDK_v1.5.4/ESP8266_NONOS_SDK`

3. Make a symbolic link
`ln -s ~/esp8266/SDK_v1.5.4/ESP8266_NONOS_SDK ~/esp8266/esp-open-sdk/sdk`

4. You're all set
`make ESP_ROOT=~/esp8266/esp-open-sdk/ all`

SinanAkkoyun, 2021-03-24:
> 1. Download ESP8266_NONOS_SDK_V1.5.4_16_05_20.zip
> `wget -O SDK_V1.5.4_16_05_20.zip https://bbs.espressif.com/download/file.php?id=1469&sid=3e67f0c75557051412485c5d4e578de7`
> 
> 2. unpack zip
> `unzip ESP8266_NONOS_SDK_V1.5.4_16_05_20.zip -d ~/esp8266/SDK_v1.5.4/ESP8266_NONOS_SDK`
> 
> 3. Make a symbolic link
> `ln -s ~/esp8266/SDK_v1.5.4/ESP8266_NONOS_SDK ~/esp8266/esp-open-sdk/sdk`
> if you would like to change the SDK reference, unlink first
> `unlink ~/esp8266/esp-open-sdk/sdk`
> 
> 4. You're all set
> `make ESP_ROOT=~/esp8266/esp-open-sdk/ all`
> 

Thank you I will try after my exams!!! ❤️❤️❤️

SinanAkkoyun, 2021-05-14:
Hi! I finally soled it all!!!!! It just took reinstalling and doing some manual symbolic links, thanks guys!

closed 2021-05-14 by SinanAkkoyun