cnlohr/espusb #61: The firmware image.elf-0x00000.bin is different when build on PC and laptop

heyzzk, 2021-12-05:
Hi,

I found this problem. Seems the content of the two firmwares on PC(thinkstaion) and laptop(thinkpad&macbook air) are the same, but the order of them are different. It's wired and I want to ensure they are the same.

Is it associated with gcc command or CPU architecture? I am not familiar with these. Could anyone give me some suggestion to ensure they are the same?

Thanks.

heyzzk, 2021-12-05:
[fw.zip](https://github.com/cnlohr/espusb/files/7655163/fw.zip)
Add the two firmwares I built on PC and laptop.

cnlohr, 2022-01-18:
It is entirely possible for there to be minor differences between output binaries.  It is a nondeterministic process.  But the output code should work across platforms.