cnlohr/espusb #62: help me to develop low-cost braille display

discapacidad5, 2021-12-06:
help with HID braille for open source devices for people with disabilities

Hello, I am developing a device for people with disabilities. I need to use the HID braille protocol but I am not getting anything with HID for esp32, could you help me with a sample code of how I can implement the HID braille protocol?

here are references to braille HID

https://usb.org/sites/default/files/hutrr78_-_creation_of_a_braille_display_usage_page_0.pdf

https://github.com/nvaccess/nvda/blob/00cd67b737bc5b23a6f5e31cf28110b64ebf2fee/devDocs/hidBrailleTechnicalNotes.md

my repository is the following
https://github.com/brailletouch/Brailletouch

discapacidad5, 2021-12-06:
error

cnlohr, 2022-01-18:
Hey, sorry, I just noticed this.  I regret that this is unrelated/unconnected with the ESP32 USB stuff.  And, I don't already have knowledge of more generic HID devices apart from what was supported in this repository :-/