cnlohr/espusb #70: How to use custom commands

mishalshanavas, 2023-05-12:
How to use custom commands 
[image]

Can anyone show an example that i can try?

cnlohr, 2023-05-12:
https://github.com/cnlohr/espusb/blob/master/user/custom_commands.c#L16

CC = custom command test

YeetTheAnson, 2024-02-24:
@cnlohr Can you give more details on it? And provide some examples

cnlohr, 2024-02-29:
If you type something in there and click submit you can send messages to the C code, I strongly recommend printing the value being passed into `CustomCommand`.  I believe you have to prefix it with `C` so for instance, `CK` would be handled in `case 'K': case 'k': //Keyboard`

Commands are separated by the tab character.  I thought it was a space, but it looks like a tab, so you might have to copy-paste a tab in, i.e. `CM0	100	100` would move the mouse by 100,100. 

See here:  https://github.com/cnlohr/espusb/blob/0a60b983234908e9cc68529544c259ca47c99b27/web/page/main.js#L59

closed 2025-09-06 by mishalshanavas