cnlohr/espusb #71: Can mqtt be used to control the mouse and keyboard?

sheminasalam, 2023-05-28:
Instead of using the browser how can I use mqtt to control the movements and clicks? 
can this project be adapted for that?

cnlohr, 2023-05-31:
Absolutely! But you would need to write the code to do it!

It wouldn't be too hard, but you would need to familiarize yourself with this codebase.

sheminasalam, 2023-05-31:
I haven't done this level of coding. can you give me som pointers on which files should I edit and how should i try to implement this. 

Also will it be possible to move the mouse to a specific point on the screen?

cnlohr, 2023-06-01:
Try to get to the point where you can printf(...) then printf a bunch of stuff in main to figure out what's actually happening.

It will be annoying and tricky to resurrect the build chain for this.   You can ask for help on my Discord server in the `espusb` channel.