cnlohr/espusb #76: Failed to compile espusb with docker

Cindy723, 2024-09-16:
Hello author，I pulled your image, and when I started docker, I was prompted ESP8266 SDK is installed, but an error was reported later:
```
In file included from esp82xx/fwsrc/commonservices.c:20:0:
esp82xx/include/spi_memory_addrs.h:31:2: error: #error FATAL ERROR: MFS_PAGE_OFFSET is 0.
 #error FATAL ERROR: MFS_PAGE_OFFSET is 0.
  ^
esp82xx/include/spi_memory_addrs.h:35:2: error: #error "end of USER_DATA doesn't line up with MFS_PAGE_OFFSET from the makefile"
 #error "end of USER_DATA doesn't line up with MFS_PAGE_OFFSET from the makefile"
  ^
esp82xx/fwsrc/commonservices.c: In function 'cmd_WiFi':
esp82xx/fwsrc/commonservices.c:499:14: error: 'SOFTAP_CHANNEL' undeclared (first use in this function)
       chan = SOFTAP_CHANNEL;

/// There are other mistakes in the back
```
How should I modify it?

Cindy723, 2024-09-16:
I made sure the hardware was working

bitsinside, 2025-01-19:
i'm getting this error, may I know what causing this error 

`~/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -mlongcalls -Os -I/home/esp/esp_iot_sdk_v1.5.2/include -Iesp82xx/include -I. -Iesp82xx/fwsrc -Iuser -DICACHE_FLASH -DVERIFY_FLASH_WRITE -DDEBUG -DFREQ=12500 -DWEB_PORT=80 -DCOM_PORT=7777 -DBACKEND_PORT=7878  -DSLOWTICK_MS=100 -DVERSSTR='"Version: 0a60b-dev - Build Sun, 19 Jan 2025 11:55:04 +0000 with -DICACHE_FLASH -DVERIFY_FLASH_WRITE -DDEBUG -DFREQ=12500 -DWEB_PORT=80 -DCOM_PORT=7777 -DBACKEND_PORT=7878  -DSLOWTICK_MS=100"'  esp82xx/fwsrc/uart.c esp82xx/fwsrc/esp82xxutil.c esp82xx/fwsrc/flash_rewriter.c esp82xx/fwsrc/http.c esp82xx/fwsrc/commonservices.c esp82xx/fwsrc/http_custom.c esp82xx/fwsrc/mdns.c esp82xx/fwsrc/mfs.c user/custom_commands.c user/user_main.c user/usb_asm_1bit.S user/usb.c -flto -Wl,--relax -Wl,--gc-sections -nostdlib -L/home/esp/esp_iot_sdk_v1.5.2/lib -L~/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/libgcc.a /home/esp/esp_iot_sdk_v1.5.2/lib/libmain.a /home/esp/esp_iot_sdk_v1.5.2/lib/liblwip.a /home/esp/esp_iot_sdk_v1.5.2/lib/libssl.a /home/esp/esp_iot_sdk_v1.5.2/lib/libupgrade.a /home/esp/esp_iot_sdk_v1.5.2/lib/libnet80211.a /home/esp/esp_iot_sdk_v1.5.2/lib/libwpa.a /home/esp/esp_iot_sdk_v1.5.2/lib/libphy.a /home/esp/esp_iot_sdk_v1.5.2/lib/liblwip.a /home/esp/esp_iot_sdk_v1.5.2/lib/libcrypto.a /home/esp/esp_iot_sdk_v1.5.2/lib/libpp.a ~/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/libgcc.a -T /home/esp/esp_iot_sdk_v1.5.2/ld/eagle.app.v6.ld -B/home/esp/esp_iot_sdk_v1.5.2/lib  -o image.elf
xtensa-lx106-elf-gcc: error: /home/esp/esp_iot_sdk_v1.5.2/lib/libmain.a: No such file or directory
xtensa-lx106-elf-gcc: error: /home/esp/esp_iot_sdk_v1.5.2/lib/liblwip.a: No such file or directory
xtensa-lx106-elf-gcc: error: /home/esp/esp_iot_sdk_v1.5.2/lib/libssl.a: No such file or directory
xtensa-lx106-elf-gcc: error: /home/esp/esp_iot_sdk_v1.5.2/lib/libupgrade.a: No such file or directory
xtensa-lx106-elf-gcc: error: /home/esp/esp_iot_sdk_v1.5.2/lib/libnet80211.a: No such file or directory
xtensa-lx106-elf-gcc: error: /home/esp/esp_iot_sdk_v1.5.2/lib/libwpa.a: No such file or directory
xtensa-lx106-elf-gcc: error: /home/esp/esp_iot_sdk_v1.5.2/lib/libphy.a: No such file or directory
xtensa-lx106-elf-gcc: error: /home/esp/esp_iot_sdk_v1.5.2/lib/liblwip.a: No such file or directory
xtensa-lx106-elf-gcc: error: /home/esp/esp_iot_sdk_v1.5.2/lib/libcrypto.a: No such file or directory
xtensa-lx106-elf-gcc: error: /home/esp/esp_iot_sdk_v1.5.2/lib/libpp.a: No such file or directory
esp82xx/main.mf:78: recipe for target 'image.elf' failed`

bitsinside, 2025-01-19:
try to hole reset and boot button together, then hit command to write firmware. Then first release reset button while holding boot button then release the boot button